platformio/platformio-vscode-ide #86: IDE keeps reinstalling if useBuiltinPIOCore set to 'false'

burillo-se, 2018-02-12:
PlatformIO keeps telling me that it's been "installed" each time after reload if i don't use builtin PlatformIO but rather want to use my own installation, with a custom path.

By the way, why is this not default or at least a "do you want to download it" message box?

ivankravets, 2018-02-12:
What is your own version? The minimum supported version by IDE is https://github.com/platformio/platformio-vscode-ide/blob/develop/src/constants.js#L14

We will release PIO 3.5.2 soon.

TheAifam5, 2018-02-16:
Also when `useDevelopmentPIOCore` is true and `useBuiltinPIOCore` is false, VS extension does nothing.
I removed the .platformio folder to reinstall completly and there is any CPU/HDD/Network usage.
It shows on the bottom left that PlatformIO IDE is installing, but also shows the notification that has been installed.

(when the options are set, shows directly after the start 2 messages - installing and installed)
[image]

Want to say that after commenting out "useBuiltinPIOCore", the first shows the message on status bar that is installing, then i noticed a .platformio folder has been created with content inside. Now dunno if is still installing or is installed because shows the same behavior like with `useBuiltinPIOCore = "false"`.

burillo-se, 2018-02-25:
@ivankravets OK so my platformio is 3.5.1. still, why, instead of telling me that i have an old version and erroring out (so that i know something is wrong, and know why), it just keeps "installing" PlatformIO?

closed 2018-03-08 by ivankravets

ivankravets, 2018-03-08:
1. See minimum required PIO Core version for IDE https://github.com/platformio/platformio-vscode-ide/blob/develop/src/constants.js#L14
2. We had a bug with PIO Core installer which was fixed in 0.13.2. Please update extension
3. If you want to use OWN PIO Core, please upgrade it to 3.5.2

commit 64983d2a (2018-03-08): Fix endless loop with installing PIO Core when `platformio-ide.useDevelopmentPIOCore` is set to `false` // Resolve #86