platformio/platformio-vscode-ide #87: Cannot create handlers for esp8266 freertos sdk

MohammadAhmad1990, 2018-02-19:
Cannot create any type of handlers for esp8266 freesrtos sdk one of the examples is below. kindly guide what i am going wrong.
src\main.c:8:1: error: unknown type name 'QueueHandle_t'

`#include <stdlib.h>
#include <string.h>
#include <stddef.h>
#include "esp_common.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/queue.h"

QueueHandle_t xQueue;

uint32 user_rf_cal_sector_set(void)
{
    flash_size_map size_map = system_get_flash_size_map();
    uint32 rf_cal_sec = 0;

    switch (size_map) {
        case FLASH_SIZE_4M_MAP_256_256:
            rf_cal_sec = 128 - 5;
            break;

        case FLASH_SIZE_8M_MAP_512_512:
            rf_cal_sec = 256 - 5;
            break;

        case FLASH_SIZE_16M_MAP_512_512:
        case FLASH_SIZE_16M_MAP_1024_1024:
            rf_cal_sec = 512 - 5;
            break;

        case FLASH_SIZE_32M_MAP_512_512:
        case FLASH_SIZE_32M_MAP_1024_1024:
            rf_cal_sec = 1024 - 5;
            break;
        case FLASH_SIZE_64M_MAP_1024_1024:
            rf_cal_sec = 2048 - 5;
            break;
        case FLASH_SIZE_128M_MAP_1024_1024:
            rf_cal_sec = 4096 - 5;
            break;
        default:
            rf_cal_sec = 0;
            break;
    }

    return rf_cal_sec;
}

void user_init()
{
xQueue =xQueueCreate(20,2);
}`

MohammadAhmad1990, 2018-02-19:
It seems I have got it working  the below way is to define handlers for queue and tasks.
xQueueHandle xQueue;
xTaskHandle xtask1;

my question is why does it differ from "Mastering FreeRtos" book

ivankravets, 2018-02-21:
https://community.platformio.org/

closed 2018-02-21 by ivankravets