platformio/platformio-vscode-ide #88: SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version

tcm0116, 2018-02-24:
Running macOS 10.12.3, Visual Studio Code 1.20.1, and PlatformIO IDE extension version  0.12.1.

After installing the PlatformIO IDE extension and reloading VSCode, the installation of PlatformIO fails with the following error:

```
PIP: Exception:
Traceback (most recent call last):
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 335, in run
    wb.build(autobuilding=True)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/wheel.py", line 749, in build
    self.requirement_set.prepare_files(self.finder)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/cachecontrol/adapter.py", line 47, in send
    resp = super(CacheControlAdapter, self).send(request, **kw)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)
```

So far as I can tell, this is being caused by an older version of OpenSSL on my system. However, I can't find a way to get the Python penv (in `~/.platformio/penv/`) installed by PlatformIO to utilize the newer version of OpenSSL (1.0.2n) and Python (2.7.14) I've installed using `brew`.

I've been using VSCode with the PlatformIO IDE extension for months, and today it just stopped working. I was getting the same TLSV1_ALERT_PROTOCOL_VERSION error when just trying to build a project I've built hundreds of times in the past. I tried completely uninstalling VSCode (and deleting all of it's support files in the system) and reinstalling from scratch, but I can't get past this error.

ivankravets, 2018-02-24:
Duplicate of https://github.com/platformio/platformio-core/issues/1390

Please re-test

ivankravets, 2018-02-26:
What is your OS?

tcm0116, 2018-02-26:
Please read the original post. All of the relevant details are provided.

ivankravets, 2018-02-26:
Could you try to remove `~/.platformio/penv/` and restart VSCode?

tcm0116, 2018-02-27:
Uninstalled PlatformIO IDE from VSCode, removed the entire `~/.platformio/` folder, and reloaded VSCode, but still got the same error.

ivankravets, 2018-02-27:
Please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

tcm0116, 2018-02-28:
Here is the output requested:

```
$ echo $PATH
/usr/local/opt/openssl/bin:/usr/local/opt/python/libexec/bin:/Users/thomas/perl5/bin:/Users/thomas/perl5/perlbrew/bin:/Users/thomas/perl5/perlbrew/perls/perl-5.16.0/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin:/usr/local/opt/openssl/bin:/usr/local/opt/python/libexec/bin:/Users/thomas/perl5/bin
$ echo $PYTHONPATH

$ python --version
Python 2.7.14
$ pip --version
pip 9.0.1 from /usr/local/lib/python2.7/site-packages (python 2.7)
$ pip list
bottle (0.12.13)
certifi (2018.1.18)
chardet (3.0.4)
click (5.1)
colorama (0.3.9)
idna (2.6)
lockfile (0.12.2)
pip (9.0.1)
platformio (3.5.1)
pyserial (3.4)
requests (2.18.4)
semantic-version (2.6.0)
setuptools (36.5.0)
urllib3 (1.22)
wheel (0.30.0)
```

Here is the output using a script that can determine the version of OpenSSL ([test.py.txt](https://github.com/platformio/platformio-vscode-ide/files/1767395/test.py.txt)):

```
$ ./test.py
Python info: 2.7.14
OpenSSL info: OpenSSL 1.0.2n  7 Dec 2017
```

However, I don't think running these commands from the system terminal provide any useful information since PlatformIO runs within the environment provided by `~/.platformio/penv/`. Here is the output after running `. ~/.platformio/penv/bin/activate` to be within the penv that PlatformIO uses within VSCode:

```
$ . ~/.platformio/penv/bin/activate
(penv) $ echo $PATH
/Users/thomas/.platformio/penv/bin:/usr/local/opt/openssl/bin:/usr/local/opt/python/libexec/bin:/Users/thomas/perl5/bin:/Users/thomas/perl5/perlbrew/bin:/Users/thomas/perl5/perlbrew/perls/perl-5.16.0/bin:/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:/usr/local/CrossPack-AVR/bin
(penv) $ echo $PYTHONPATH

(penv) $ python --version
Python 2.7.10
(penv) $ pip --version
pip 9.0.1 from /Users/thomas/.platformio/penv/lib/python2.7/site-packages (python 2.7)
(penv) $ pip list
pip (9.0.1)
setuptools (38.5.1)
wheel (0.30.0)
```

And here is the output from the test script showing the OpenSSL version:

```
(penv) $ ./test.py
Python info: 2.7.10
OpenSSL info: OpenSSL 0.9.8zh 14 Jan 2016
```

It is known that OpenSSL 0.9.8zh only supports TLSv1. On macOS, OpenSSL 0.9.8zh is the only available version provided by the system. As such, I think the way to fix this issue is to include a newer version of OopenSSL in the penv used by PlatformIO.

ivankravets, 2018-02-28:
Could you try to install `pyopenssl` in `penv`?
```
(penv) pip install pyopenssl platformio 
(penv) platformio upgrade -–dev
```

tcm0116, 2018-03-01:
I got the following error:

```
(penv) $ platformio upgrade --dev
Please wait while upgrading PlatformIO ...
Error: ['/Users/thomas/.platformio/penv/bin/python', '-m', 'pip', '--no-cache-dir', 'install', '--upgrade', 'https://github.com/platformio/platformio-core/archive/develop.zip']
Collecting https://github.com/platformio/platformio-core/archive/develop.zip

Exception:
Traceback (most recent call last):
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/commands/install.py", line 324, in run
    requirement_set.prepare_files(finder)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 380, in prepare_files
    ignore_dependencies=self.ignore_dependencies))
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/req/req_set.py", line 620, in _prepare_file
    session=self.session, hashes=hashes)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 821, in unpack_url
    hashes=hashes
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 659, in unpack_http_url
    hashes)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 853, in _download_http_url
    stream=True,
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 488, in get
    return self.request('GET', url, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/download.py", line 386, in request
    return super(PipSession, self).request(method, url, *args, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 475, in request
    resp = self.send(prep, **send_kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/sessions.py", line 596, in send
    r = adapter.send(request, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/pip/_vendor/requests/adapters.py", line 497, in send
    raise SSLError(e, request=request)
SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)

* Upgrade using `pip install -U platformio`
* Try different installation/upgrading steps:
  http://docs.platformio.org/page/installation.html
```

Running my test script prints the following, despite the fact that pyOpenSSL has been installed:

```
(penv) $ ~/test.py 
Python info: 2.7.10
OpenSSL info: OpenSSL 0.9.8zh 14 Jan 2016
```

Shuurak, 2018-03-01:
Баг тот же, но при изначальной установке на VSCode

`SSLError: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:661)`

OSX 10.9.5
VS 1.20.1
PIO 3.5.1
OpenSSL 1.0.2n

ivankravets, 2018-03-02:
Please open system terminal and provide an output of 
```
python -c "import ssl; print(ssl.OPENSSL_VERSION)"
```

ivankravets, 2018-03-02:
Try in system terminal
```
/usr/bin/ruby -e "$(curl -fsSL https://raw.githubusercontent.com/Homebrew/install/master/install)"
brew update
brew install openssl
```

After that, please remove `~/.platformio/penv` folder and restart IDE.

tcm0116, 2018-03-02:
This isn't going to work. I can install whatever I want to the system using brew or whatever, but the penv provided by PlatformIO will use the default system-provided version of OpenSSL, which is 0.9.8zh, which only supports TLSv1. I suspect the only way to get this to work will be for PlatformIO to provide an updated version of OpenSSL as part of the penv.

```
$ python -c "import ssl; print(ssl.OPENSSL_VERSION)"
OpenSSL 1.0.2n  7 Dec 2017
$ . ~/.platformio/penv/bin/activate
(penv) $ python -c "import ssl; print(ssl.OPENSSL_VERSION)"
OpenSSL 0.9.8zh 14 Jan 2016
$ deactivate
$ brew install openssl
Warning: openssl 1.0.2n is already installed
$ rm -rf ~/.platformio/penv
```

After restarting PlatformIO, I still get the same error as in the original post.

ivankravets, 2018-03-02:
Do you use the same Pythons? Please check `python --version` in (penv) and outside it.

tcm0116, 2018-03-02:
That infirmation was provided in https://github.com/platformio/platformio-vscode-ide/issues/88#issuecomment-369256877

closed 2018-03-03 by ivankravets

ivankravets, 2018-03-03:
Please upgrade PlatformIO IDE for VSCode to 0.13.0

tcm0116, 2018-03-03:
I was able to get PlatformIO installed into VSCode. However, I'm still having a few issues. First, I get a "PlatformIO: IntelliSense Index failed: Error: Traceback (most recent call last):" popup every few seconds. The traceback is not actually displayed, and I don't know how to see it. I then tries to run the Rebuild IntelliSense Index task, but got the following error in the terminal:

```
> Executing task: platformio init --ide vscode <

The current working directory /Users/thomas/Documents/3DPrinter/Marlin will be used for project.
You can specify another project directory via
`platformio init -d %PATH_TO_THE_PROJECT_DIR%` command.

The next files/directories have been created in /Users/thomas/Documents/3DPrinter/Marlin
platformio.ini - Project Configuration File
src - Put your source files here
lib - Put here project specific (private) libraries
Error: Traceback (most recent call last):
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/__main__.py", line 120, in main
    cli(None, None, None)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 680, in main
    rv = self.invoke(ctx)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/click/decorators.py", line 16, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/commands/init.py", line 102, in cli
    pg = ProjectGenerator(project_dir, ide, env_name)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 35, in __init__
    self._gather_tplvars()
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 152, in _gather_tplvars
    self._tplvars.update(self.get_project_build_data())
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/util.py", line 134, in __call__
    value = self.func(*args)
  File "/Users/thomas/.platformio/penv/lib/python2.7/site-packages/platformio/ide/projectgenerator.py", line 76, in get_project_build_data
    raise result.exception
SSLError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /teemuatlut/TMC2208Stepper/archive/v0.0.3.zip (Caused by SSLError(SSLError(1, u'[SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)'),))

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================
```

ivankravets, 2018-03-05:
@tcm0116 this problem is linked with incorrect Python's version which PlatformIO chose for the virtual environment. We look for the first compatible Python interpreter and use it. In your case, `/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/` system's Python (2.7.10) is located before "brew"s. Please update your `~/.bash_profile`. For example, add these lines at the top of file:

```
PATH="/usr/local/bin:/usr/local/sbin:${PATH}"
export PATH
```

After that, please remove `~/.platformio/penv` folder and restart VSCode.

tcm0116, 2018-03-05:
Thanks. I will give that a try this evening. 

Maybe it works help if the compatibility check was extended to either get the highest version available, or one that supports TLS better than v1 due to GitHub's new requirements.

tcm0116, 2018-03-07:
My `~/.bash_profile` already contained the following:

`export PATH="/usr/local/opt/openssl/bin:/usr/local/opt/python/libexec/bin:$PATH"`

As such, it appears that the python version is not being found in the order you're expecting.

ivankravets, 2018-03-07:
You need to place brew's Python before a system Python.

```
export PATH="/usr/local/bin:/usr/local/sbin:$PATH"
```

tcm0116, 2018-03-07:
>You need to place brew's Python before a system Python.

It is. On my system, brew installed Python at `/usr/local/opt/python/libexec/bin`.

ivankravets, 2018-03-07:
What is your default shell? Do you have multiple shells in a system? It seems that VSCode starts with another shell where PATH differs.

tcm0116, 2018-03-07:
I tried this before when I couldn't get PlatformIO installed at all, but maybe it'll work this time:

https://apple.stackexchange.com/questions/51677/how-to-set-path-for-finder-launched-applications

However, even if it does work, this is way too much to ask of the average user to have to install brew and change all of these paths.

ivankravets, 2018-03-07:
> However, even if it does work, this is way too much to ask of the average user to have to install brew and change all of these paths.

You are the first with this issue :( I use `brew` too and don't have any problems. I used their generic installer and the working BIN directory is located in `/usr/local/bin`.

Also, you can create virtual environment manually with own Python:

```
pip install virtualenv
rm -rf ~/.platformio/penv
virtualenv ~/.platformio/penv -p /path/to/custom/python
~/.platformio/penv/bin/activate
(penv) pip install platformio
(penv) pio upgrade --dev
```

DrewParallel, 2018-04-13:
I have the same issue. Steps above did not resolve the problem.

` pip install platformio`

`Collecting platformio
  Could not fetch URL https://pypi.python.org/simple/platformio/: There was a problem confirming the ssl certificate: [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:661) - skipping
  Could not find a version that satisfies the requirement platformio (from versions: )
No matching distribution found for platformio`

aasuuu, 2018-04-16:
I have solved the issue by performing these steps

Download get-pip.py from https://bootstrap.pypa.io/get-pip.py
Launch terminal or command prompt 
cd to Downloads folder or where the get-pip.py is downloaded 

Run command 
python get-pip.py

And Yo, every installation is working now 
Njoy 

Ashutosh Gupta

gitanupam, 2018-04-25:
Upgrade `pip` without using `pip`. See this SO post: https://stackoverflow.com/q/49768770/1526703

chanckstephen, 2018-06-03:
Hi,

I'm a newbie. I'm not using Platform IO but I'm using Kivy. I suffer from the same error during Buildozer (v0.35dev) compilation to produce an Android APK (my Kivy program works fine on the desktop):
IOError: [Errno socket error] [SSL: TLSV1_ALERT_PROTOCOL_VERSION] tlsv1 alert protocol version (_ssl.c:590)

My Kivy (v1.10.0) is running on Python 2.7.10 and the OpenSSL version reported is OpenSSL 0.9.8zh 14 Jan 2016. My OS is 10.12.6.

Can someone please teach me how to upgrade Kivy's OpenSSL to a later version so that the above error can be solved?

Thanks in advance.

-Stephen

mkaykisiz, 2018-06-03:
curl https://bootstrap.pypa.io/get-pip.py | python

chanckstephen, 2018-06-04:
I managed to upgrade my Mac's python's (2.7.15) openssl to OpenSSL 1.0.2o  27 Mar 2018, but Kivy's python (2.7.10) is another version whose OpenSSL is OpenSSL 0.9.8zh 14 Jan 2016. I'm trying to update THIS version of OpenSSL.

commit e54f1afa (2018-03-03): Fix GitHub's "TLSV1_ALERT_PROTOCOL_VERSION" issue with PIO Core installer // Resolve #88