platformio/platformio-vscode-ide #89: VS Code and tasks

whoisvolos, 2018-02-24:
Good day!

When I've tried to edit monitor task in VS Code to change baud rate to 115200 (my board does not want to work @ 9600 baud): 

Was:
`{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "args": [
                "device",
                "monitor"
            ],
            "problemMatcher": [
                "$platformio"
            ]
        }
    ]
}`

Became:
`{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "args": [
                "device",
                "monitor",
                "-b",
                "115200"
            ],
            "problemMatcher": [
                "$platformio"
            ]
        }
    ]
}`

After change I've tried to run monitor task again. Serial monitor still worked on 9600. When I've opened task definiton again, I've got that task definition is doubled, like this:

`{
    // See https://go.microsoft.com/fwlink/?LinkId=733558
    // for the documentation about the tasks.json format
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "args": [
                "device",
                "monitor",
                "-p",
                "115200"
            ],
            "problemMatcher": [
                "$platformio"
            ]
        },
        {
            "type": "PlatformIO",
            "args": [
                "device",
                "monitor"
            ],
            "problemMatcher": [
                "$platformio"
            ]
        }
    ]
}`

Monitor still works on 9600.  Is there a chance to work with serial monitor from VS Code + PlatformIO IDE on baudrates different from 9600? Of I can run needed command in VSC terminal / powershell instance and so on.

W10 x64 / Python 2.7.14 / VS Code 1.20.1 / PIO 3.5.2rc1 (IDE version, ./penv/Scripts/ added to PATH)

ivankravets, 2018-03-03:
I'll take a look at it soon. Temporary solution

* http://docs.platformio.org/en/latest/ide/vscode.html#serial-port-monitor

ivankravets, 2018-05-26:
@dbaeumer  What is wrong with our code? https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L230

**tasks.json**
```js
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "label": "Custom Monitor",
            "args": [
                "device",
                "monitor",
                "-b",
                "115200"
            ],
            "problemMatcher": [
                "$platformio"
            ]
        }
    ]
}
```

I can reproduce this issue, VSCode errors with
```
Error: The PlatformIO task detection didn't contribute a task for the following configuration:
{
    "type": "PlatformIO",
    "label": "Custom Monitor",
    "args": [
        "device",
        "monitor",
        "-b",
        "115200"
    ],
    "problemMatcher": [
        "$platformio"
    ]
}
The task will be ignored.

```

dbaeumer, 2018-05-28:
@ivankravets that usually means that the extension contributing the task didn't get activated hence the task customization in tasks.json can't be mapped to any provided task. So we ignore it. 

Does the extension specify activation event

```
  "activationEvents": [
    "onCommand:workbench.action.tasks.runTask"
  ]
```

ivankravets, 2018-05-28:
@dbaeumer see https://github.com/platformio/platformio-vscode-ide/blob/4f1ac73014aeea4a6f616cdf6ebc0310c87d3dc4/package.json#L38

dbaeumer, 2018-05-30:
OK. I see. How does the `taskDefinitions` contribution in the package.json for the `PlatformIO` type look like.

ivankravets, 2018-05-30:
Do you mean this https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L230:L254 ?

dbaeumer, 2018-05-30:
No, for each task type you need to define a `taskDefinitions` which define the attributes to compare tasks. Otherwise we can't map a tasks from the tasks.json to a task provided by a task provider. Here is an example for gulp: https://github.com/Microsoft/vscode/blob/master/extensions/gulp/package.json#L46

dbaeumer, 2018-05-30:
The args in the tasks.json must match the args in the provided task. Since it has `115200` could it be that the contributed task has a different number. Or is this constant.

ivankravets, 2018-05-30:
Now I have the question: "What does mean ⚙️  "Configure task" button near each of PIO task?". How can people change default arguments? Is it possible with the current task API?
```js
    const task = new vscode.Task(
      {
        type: PIOTasksProvider.title,
        args: this._args
      },
      this.name,
      PIOTasksProvider.title,
      new vscode.ProcessExecution(IS_WINDOWS ? 'platformio.exe' : 'platformio', this._args, {
        env: process.env
      }),
      '$platformio'
    );
```

The idea of this issue is to re-use some configuration and just tweak arguments.

ivankravets, 2018-05-30:
Ok, it seems that it was a mistake on my side. I've just made some changes and it still does not work:

# Task defination

```js
    "taskDefinitions": [
      {
        "type": "PlatformIO",
        "required": [
          "task"
        ],
        "properties": {
          "task": {
            "type": "string",
            "description": "Task name"
          }
        }
      }
    ],
```

# Task provider

```js
    const task = new vscode.Task(
      {
        type: PIOTasksProvider.title,
        task: this.name
      },
      this.name,
      PIOTasksProvider.title,
      new vscode.ShellExecution(IS_WINDOWS ? 'platformio.exe' : 'platformio', this._args, {
        env: process.env
      }),
      '$platformio'
    );
```

# Custom task in `tasks.json`
```js
{
    "version": "2.0.0",
    "tasks": [
        {
            "type": "PlatformIO",
            "task": "Monitor",
            "problemMatcher": [
                "$platformio"
            ],
            "options": {
                "shell": {
                    "args": [
                        "unknown_arg"
                    ]
                }
            },
            "label": "PlatformIO: Custom Monitor"
        }
    ]
}
```

# Result

It still runs original "Monitor" task and does not fail with `"unknown_arg"`.

Also, could we use `vscode.ProcessExecution` instead of `vscode.ShellExecution` and make it configurable from `tasks.json`?

dbaeumer, 2018-05-30:
They can tweak everything except the attributes that defines the `identifier` (or `key` if you want) of the tasks. The attributes in the `taskDefinitions` define the task identity. If they are different then VS Code things they are two different task.

So if you want to let users tweak the args they shouldn't be part of the identity.

ivankravets, 2018-05-30:
Now they are similar. Why it does not work? The same "type" and the same "task" for TaskDefination.

dbaeumer, 2018-05-30:
Do you still get the warning that the task in tasks.json can not be found `Error: The PlatformIO task detection didn't contribute a task for the following configuration` ?

If not they are still mapped but I am not sure why the unknown_arg doesn't make it fail.

Yes, you can customize process tasks the same way as you can customize shell tasks.

ivankravets, 2018-05-30:
1) No warnings
2) I even see this custom task in a menu
3) See result below, it now shows that ran a "Custom task" but arguments are still from the original task.

[image]

dbaeumer, 2018-05-30:
We only respect different shell args if the shell literal also defines an executable :-). I can change this but usually it doesn't make sense to provide shell args without an executable. 

Just to clarify: the shell args are the args passed to the shell process used to execute the command. They have nothing to do with args passed to the command. For example if you want to use bash to execute the command the shell args are `-c` in PowerShell it is `-Command`

dbaeumer, 2018-05-30:
`options.shell` control the shell interpreter to be used.

ivankravets, 2018-05-30:
> I can change this but usually it doesn't make sense to provide shell args without an executable.

1) It does. The great use-case with this custom task that you don't need  to customize `executable` file. This is dynamic path to PIO Core which depends on a system. Somewehre it is installed globally, somewhere it will be FULL path to binary
2) I like the idea where we can configure separately EXECUTABLE and ARGS. User can use own version of PIO Core and keep default (our) pre-configred arguemnts.

Is it difficult to make it working as I described in example above? I'll update our docs for this use case.

Thanks!

ivankravets, 2018-05-30:
@dbaeumer see https://github.com/platformio/platformio-vscode-ide/commit/4218897b6bacd4a2740a11844a52e8160bb2341e

Do we need something more on our side to close this issue?

dbaeumer, 2018-05-30:
Regarding https://github.com/platformio/platformio-vscode-ide/issues/89#issuecomment-393172980. The change looks good to me.

ivankravets, 2018-05-30:
> Do your users really need to customize the args for the command interpreter ?

Yes, PlatformIO for VSCode is just a wrapper around PIO Core (CLI). Only 20-30% of PIO Core are covered in VSCode now :(

ivankravets, 2018-05-30:
For example, a default value for `PlatformIO: Monitor` task is `$ platformio device monitor`. Nevertheless, take a look at the whole list of available options here http://docs.platformio.org/en/latest/userguide/cmd_device.html#platformio-device-monitor

ivankravets, 2018-05-30:
@dbaeumer done! https://github.com/Microsoft/vscode/issues/50819

ivankravets, 2018-06-20:
@whoisvolos Please take a look at the updated docs http://docs.platformio.org/en/latest/ide/vscode.html#custom-tasks

As @dbaeumer mentioned, we can not modify arguments of original PlatformIO's task. The shell's type task should be used instead.

closed 2018-06-20 by ivankravets

Leandro-G-Ribeiro, 2018-12-15:
the simple task of changing the terminal speed of the platformio in vscode can be accomplished by changing the defaut option in the same terminal python file. I had this problem in changing the default speed of the serial terminal in vscode and I could not with any tips found on the internet, I just managed to change the default value in the same terminal file. But it's quite simple: just find the Device file in the path: C: \ Users \ name user \ .platformio \ penv \ Lib \ site-packages \ platformio \ commands and modify the line: "default_baudrate = kwargs ['baud'] or 9600, "for the amount you want. This file must be opened with the IDLE of python or in the same vscode. Edit and save this file to take effect. Once you have done this you already have a default value that you want in your terminal. It's so much easier than many tutorials. is not the most correct form but while vscode does not offer a more practical way to do this as ATOM does, this is the most practical way.

ivankravets, 2018-12-15:
https://docs.platformio.org/en/latest/projectconf/section_env_monitor.html

commit ceb85ee0 (2018-06-20): Custom Tasks // Resolve #89