platformio/platformio-vscode-ide #91: In VS Code: Unable to open 'PIO Home': Cannot read property 'theme' of undefined.

jetpathar, 2018-03-03:
After a fresh install of VS Code and Plaformio-IDE on W10,
when clicking on the PlatformIO Home button I get:

"Unable to open 'PIO Home': Cannot read property 'theme' of undefined."

I tried to uninstall, reboot and reinstall several times, still the same...

viordash, 2018-03-03:
Temporary workaround:
- in \.vscode\extensions\platformio.platformio-ide-0.13.0\lib\home.js
replace lines,
```
 return 
      <html>
      <body style="margin: 0; padding: 0; height: 100%; overflow: hidden; background-color: ${theme === 'light' ? '#FFF' : '#1E1E1E'}">
        <iframe src="${pioNodeHelpers.home.getFrontendUri(params.host, params.port, { start, theme })}"
          width="100%"
          height="100%"
          frameborder="0"
          style="border: 0; left: 0; right: 0; bottom: 0; top: 0; position:absolute;" />
      </body>
      </html>
    ;
```

with lines from [previous version](https://github.com/platformio/platformio-vscode-ide/blob/e54f1afa485a496d5e3da19d0d1b785fa9df18ab/src/home.js) of this file

    return `
      <html>
      <body style="margin: 0; padding: 0; height: 100%; overflow: hidden; background-color: #fff">
        <iframe src="http://${ params.host }:${ params.port}?start=/${ uri.authority }" width="100%" height="100%" frameborder="0" style="border: 0; left: 0; right: 0; bottom: 0; top: 0; position:absolute;" />
      </body>
      </html>
    `;

Then restart vscode.

jetpathar, 2018-03-03:
Hi viordash,
Thank you so much for your fast reply, it works, I search on the web for one hour
and I didn't find anything about my problem and you fixed it right away !
Cheers !

viordash, 2018-03-04:
In my case, after returning back the changes also continue to work. Apparently something produced registration and the system works even after reinstallation. Tested on Win 8.1 and Win 10

VladimirAkopyan, 2018-03-04:
Same problem

rselec, 2018-03-04:
same problem here,

workaround works:

I copied the file home.js (you find it in the user directory)
Modified the lines mentioned above
started Visual Studio Code
PIO Home launches
closed Visual Studio Code
copied back the home.js file (revert any changes)
Visual Studio Code is running anyway

Please fix asap, getting started at clean installation with crazy error will made almost all people switch to another IDE...

dylanlrrb, 2018-03-04:
Im having the same problem and want to try this fix but I'm unable to find the home.js file to change it. How did you open it?

ivankravets, 2018-03-05:
Sory for the issue. We've just fixed it quickly and released 0.13.1. Please update the extension.

ivankravets, 2018-03-05:
@dylanlrrb does it work for you now?

closed 2018-03-05 by ivankravets

Super169, 2018-03-12:
I have followed above instruction except that my platformIO folder is 0.13.2.
But unfortunately,  the home page still cannot be opened.  There has no error but keep showing the blank screen for loading.
I'm working behind proxy, and login is required.  I have entered the proxy login when vs.code started, is there any other changes required?   But it works fine before the upgrade.

ivankravets, 2018-03-12:
@Super169 could you try to open PIO IDE Terminal and type `pio home --port 8888`. Do you see any errors?

alexgalsyan, 2018-03-23:
Hello I have same problem in 0.14.0 version, and I have lost terminal

ivankravets, 2018-03-23:
@alexgalsyan VSCode > Tasks > Run > PlatformIO: Upgrade PIO Core. Retsart IDE.

Or, remove `~/.platformio/packages` folder and restart IDE.

alexgalsyan, 2018-03-23:
I hade "remove ~/.platformio/packages folder and restart IDE.". this not help

ivankravets, 2018-03-23:
@alexgalsyan Please open PIO IDE Terminal and provide `pio --version`.

alexgalsyan, 2018-03-23:
how can I open ?! I have lost my Platform IO terminal shortcuts

alexgalsyan, 2018-03-23:
version of Platform IO IDE  0.14.0

ivankravets, 2018-03-23:
VSCode > View > Command Palette > Terminal

alexgalsyan, 2018-03-23:
command 'platformio-ide.newTerminal' not found

ivankravets, 2018-03-23:
[image]

alexgalsyan, 2018-03-23:
PS C:\Users\G_Aleksandr> pio --version
pio : The term 'pio' is not recognized as the name of a cmdlet, function, script file, or operable program. Check the spelling of the name, or if a path was included, ve
rify that the path is correct and try again.
At line:1 char:1
+ pio --version
+ ~~~
    + CategoryInfo          : ObjectNotFound: (pio:String) [], CommandNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

ivankravets, 2018-03-23:
Please remove `C:\Users\G_Aleksandr\.platformio` folder and restart IDE. Does it work now? It seems that you have some problems with VSCode/Extension.

alexgalsyan, 2018-03-23:
I have Delete .platformio folder and vscode/extensions platformio.platformio-ide-0.14.0 after reinstall 
reopen IDE
Platform IO IDE from VSCode extension add again
Problems same, if you need fix this issue I can gave you my TeamViewer ID

ivankravets, 2018-03-23:
>  I can gave you my TeamViewer ID

Please, me@ikravets.com

Super169, 2018-03-26:
Sorry I have missed you suggestion above.
The problem has been solved after I have uninstalled PlatformIO IDE, and delete the .PlatformIO folder, then reinstall PlatformIO IDE.
But there has problem whenever there has update on PlatformIO, and I have reinstalled PlatformIO IDE again today due to update issue.

commit a5324a7b (2018-03-05): Fix "Cannot read property 'theme' of undefined" when opening PIO Home // Resolve #91