platformio/platformio-vscode-ide #95: disable *all* PlatformIO update checks

benswift, 2018-03-16:
Is there a config setting (or something) to tell the PlatformIO VSCode IDE to never check for updates *at all*?

The situation is that I'm forced (for reasons beyond my control) to mount the `~/.platformio/packages/` and `~/.platformio/platforms` directories somewhere which is read-only. This is ok - I can get the contents of those directories and mount them, but whenever platformio goes to check and update anything in `~/.platformio` (e.g. even `appstate.json`) it fails with a "permission denied".

I know that what I'm trying to do is a *bit* off the beaten path, and so if there's no easy way to do it I understand, but I'm interested to know if there's some setting (either in the VSCode IDE or pio-core) to make it just completely satisfied with what's currently in the `./platformio` directory - never to try and change it in any way.

I'm on Ubuntu 17.10, using PlatformIO VSCode IDE. Non-portable solutions are ok.

Thanks :)

ivankravets, 2018-04-05:
http://docs.platformio.org/en/latest/userguide/cmd_settings.html#settings

closed 2018-04-05 by ivankravets

benswift, 2018-04-07:
Great, thanks.