platformio/platformio-vscode-ide #96: Can't install IDE from behind corporate firewall

LNRawz, 2018-03-16:
Hello, I am getting this error:
`Error: Failed to download file https://github.com/platformio/platformio/archive/develop.zip: Error: self signed certificate in certificate chain`
And then the OUTPUT on VSCode is:
`Installing PlatformIO Core...
Please don't close this window and don't open other folders until this process is completed.
Failed to install PlatformIO IDE.`

VSCode setting with this: "http.proxyStrictSSL": false

Firewall here (proxy engine) changes the certificate from host (this case github) to serve me. (So it can peek inside the package, darn big brother...)

amamabel, 2018-05-04:
Same issue here in Atom. Some research shows that the module platform-node-helpers gets there own python/pip environment via the nodejs request functions. I think this calls must be modified to ignored the strict ssl or add a custom cacert.pem file? Weird stuff, why platformio don't use the users pip environment or get at least a user config.

jankap, 2018-06-19:
Any news on that? I'm not able to install platformIO behind a proxy, too.

ivankravets, 2018-06-19:
Could you try this solution https://stackoverflow.com/questions/25981703/pip-install-fails-with-connection-error-ssl-certificate-verify-failed-certi?

Create `pip.ini` and add `github.com` to `trusted-host`

jankap, 2018-06-22:
It's working now after doing:

- setting `http.proxyStrictSSL = false` in VS Code
- exporting the root CA from my corporate firewall (zscaler), using chrome and converting it to .pem format, see https://stackoverflow.com/questions/31205438/docker-on-windows-boot2docker-certificate-signed-by-unknown-authority-error/36454369#36454369 
- creating a pip.ini (%home%/pip/pip.ini) with
```    
[global] 
cert = C:\Users\username\.ssh\myRootCa.pem
proxy = proxyAddress:port [without http or https]
```
- made sure, no addition python installation or at least environment variables are exported
- created environment variables `HTTP_PROXY` and `HTTPS_PROXY` 

I did not try out to add certain URLs to `trusted-host`, because I don't want to screw up my setup again, sorry. Additionally I'm not sure wich steps were necessary, though without the cacert it didn't work.

Only problem left was `tool-scons could not be found`, I had to install it manually from https://bintray.com/platformio/dl-packages/tool-scons  to ~/.platformio/packages/tool-scons

I'm on Windows 7, behind a zscaler proxy/firewall which does some kind of man in the middle ssl "attack" to scan ssl traffic. 
Maybe this helps somebody :)

Thanks,
Jan

amamabel, 2018-06-22:
Thanks for sharing Jan. I will try it.

ivankravets, 2018-10-30:
Does this solution work for you https://community.platformio.org/t/platformio-ide-in-vscode-cant-connect-to-the-internet-blocked-by-proxy/5449/3?u=ivankravets?

If yes, we will update our docs.

Thanks!

sclarke81, 2018-11-02:
I think it does. I initially added the trailing slash to my HTTP and HTTPS environment variables but it didn't work. However, I then decided to have a clear out; I uninstalled Python, uninstalled the extension, deleted the extension folder, deleted the Python folder, removed my pip ini file. My current setup is:

- `http.proxyStrictSSL = false` set in VS Code
- environment variables `HTTP_PROXY` and `HTTPS_PROXY` set to `http://<proxyaddress>:<port>/
- Python 2.7.15 x86

I've got Win 10 x64 and I'm not sure if I should use Python x86 or x64?

Now I consistently get `Error: Could not create PIO Core Virtual Environment` which is reported as issue #154 so I'll switch to there to try to resolve that.

ivankravets, 2018-11-05:
Could you open VSCode > Menu > Help > Toggle Developers Tools > Console. Do you see any errors? Please share here.

sclarke81, 2018-11-07:
I've noticed that it alternates between two different errors each time I open VSCode.

Log 1:
```
/C:/Users/<username>/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4274 Overwriting grammar scope name to file mapping for scope source.fsharp.
Old grammar file: file:///c%3A/Users/<username>/AppData/Local/Programs/Microsoft%20VS%20Code/resources/app/extensions/fsharp/syntaxes/fsharp.tmLanguage.json.
New grammar file: file:///c%3A/Users/<username>/.vscode/extensions/ionide.ionide-fsharp-3.29.0/syntaxes/fsharp.json
e.register @ /C:/Users/<username>/AppData/Local/Programs/Microsoft VS Code/resources/app/out/vs/workbench/workbench.main.js:4274
console.ts:136 [Extension Host] Git extension API method 'getGitPath' is deprecated.
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
console.ts:136 [Extension Host] Error: 'platformio' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn platformio ENOENT
	at C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:1:18484
	at y (C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:1:13369)
	at ChildProcess.<anonymous> (C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:1:13132)
	at emitOne (events.js:116:13)
	at ChildProcess.emit (events.js:211:7)
	at ChildProcess.cp.emit (C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\cross-spawn\lib\enoent.js:30:37)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:198:12)
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
console.ts:136 [Extension Host] Error: User's Virtualenv: 
	at C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:14:24678
	at ChildProcess.y (C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\platformio-node-helpers\lib\index.js:1:13369)
	at emitTwo (events.js:126:13)
	at ChildProcess.emit (events.js:214:7)
	at ChildProcess.cp.emit (C:\opt\.vscode\extensions\platformio.platformio-ide-1.4.3\node_modules\cross-spawn\lib\enoent.js:34:29)
	at maybeClose (internal/child_process.js:925:16)
	at Process.ChildProcess._handle.onexit (internal/child_process.js:209:5)
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
console.ts:136 [Extension Host] Error: EPERM: operation not permitted, open 'C:\Users\<username>\.platformio\.cache\virtualenv.tar.gz'
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
console.ts:136 [Extension Host] Error: EPERM: operation not permitted, open 'C:\Users\<username>\.platformio\.cache\virtualenv.tar.gz'
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
console.ts:136 [Extension Host] Error: EPERM: operation not permitted, open 'C:\Users\<username>\.platformio\.cache\virtualenv.tar.gz'
[... 183 lines cut ...]
e.$showMessage @ mainThreadMessageService.ts:39
t._doInvokeHandler @ rpcProtocol.ts:442
t._invokeHandler @ rpcProtocol.ts:427
t._receiveRequest @ rpcProtocol.ts:344
t._receiveOneMessage @ rpcProtocol.ts:274
(anonymous) @ rpcProtocol.ts:149
e.fire @ event.ts:140
a @ ipc.net.ts:105
n._socketDataListener @ ipc.net.ts:134
emitOne @ events.js:116
emit @ events.js:211
addChunk @ _stream_readable.js:263
readableAddChunk @ _stream_readable.js:250
Readable.push @ _stream_readable.js:208
onread @ net.js:594
console.ts:136 [Extension Host] rejected promise not handled within 1 second
t.log @ console.ts:136
t._logExtensionHostMessage @ extensionHost.ts:438
(anonymous) @ extensionHost.ts:246
emitTwo @ events.js:126
emit @ events.js:214
emit @ internal/child_process.js:772
_combinedTickCallback @ internal/process/next_tick.js:141
_tickCallback @ internal/process/next_tick.js:180
```

ivankravets, 2018-11-29:
Please update to the latest 1.5.0 release and restart VSCode. Does it work now?

- https://github.com/platformio/platformio-vscode-ide/releases/tag/v1.5.0

sclarke81, 2018-11-30:
Success! PIO is installing now, but when it tries to open PIO Home I get the `Detected unknown package 'tool-pioplus'` error. I've commented on #347 with details as that seemed the most appropriate place.

Which of the proxy settings are required? Would you like me to test your suggested settings so that you can add it to your documentation?

ivankravets, 2018-11-30:
Yes, please set environment variables as described here https://community.platformio.org/t/platformio-ide-in-vscode-cant-connect-to-the-internet-blocked-by-proxy/5449/3?u=ivankravets

ivankravets, 2019-07-16:
http://docs.platformio.org/en/latest/ide/vscode.html#proxy-server-support

The latest version of PIO IDE for VSCode should work with proxy settings declared in VSCode.

closed 2019-07-16 by ivankravets