platformio/platformio-vscode-ide #97: Build run forever (vscode insiders 1.22.0)

jbeaudoin11, 2018-03-17:
Hi !

Every time i try to build in vscode insiders 1.22.0 the build is just running forever like it's frozen :
[image]

I can't even ctrl+c to end it.. it doesn't response..  
Tried with the latest stable build of vscode 1.21.1 and it works as expected.

My version of vscode :
[image]

chenjingyuanku, 2018-04-06:
I have the same problem as you。

jbeaudoin11, 2018-04-06:
Actually it's every task.. not just build.
But yeah it's pretty annoying i need to use vs code 1.21 instead

Super169, 2018-04-06:
I have the same issue, it has generated a wrong command for build & upload

If I enter "**platformio run**" in terminal, it can build the project correctly.  So the installation of platformio is correct.

But when I press  Ctrl-Alt-B, it will generate the command as "**<source path>\platformio.exe run**", which cannot build the project.

jbeaudoin11, 2018-04-06:
Yeah i noticed that !

cyberman54, 2018-04-07:
same issue here.

ivankravets, 2018-04-07:
See updated comment https://github.com/platformio/platformio-vscode-ide/issues/97#issue-306200533 for a temporary solution.

dbaeumer, 2018-04-09:
@ivankravets Sorry for that. I investigated and I know what broke it. See https://github.com/Microsoft/vscode/issues/47350#issuecomment-379661343

I will work on a fix and will add the fix to our next recovery build. There is something in the Platform IO extension you can do to work around this bug which brings the fix faster to our customers. Simply make the path to the platformio.exe absolute (only needed under Windows). Then the `findExecutable` code used to find a executable on a Path under Windows is not executed (under Linux and Mac the code never executed since the OS finds an executable on the path). On line https://github.com/platformio/platformio-vscode-ide/blob/develop/src/tasks.js#L261 `path.join` `platformio.exe` with the install location of Platform IO which on my machine is `C:\Users\dirkb\.platformio\penv\Scripts`

I apologize for the swirl that this is causing.

dbaeumer, 2018-04-09:
Alternatively the users of the Platform IO extension could add the path to the 'Scripts' to their path environment variable. 

I started to debug why this is broken and the problem is that I when VS Code tries to find the executable the `C:\Users\dirkb\.platformio\penv\Scripts` is not on the PATH in my system (e.g. the renderer) but it is in the terminal. Need to understand why this is the case.

dbaeumer, 2018-04-09:
I delivered a fix to master which will be included in the next insider build. I will also propose the fix to be included in the next recovery build.

dbaeumer, 2018-04-09:
We decided to include the fix into the recovery build. However I still recommend that the Platfrom IO extension makes the path absolute since it might take a couple of days before the recovery build is released.

closed 2018-04-09 by ivankravets

ivankravets, 2018-04-10:
Please upgrade to v0.14.1. The temporary workaround was added.

dbaeumer, 2018-04-10:
@ivankravets thanks for releasing a workaround. Appreciated!

commit 3648a05c (2018-04-09): Temporary workaround for urgent VSCode bug in v1.22 with a broken task runner for Windows OS // Resolve #97