platformio/platformio-vscode-ide #107: Upload and other buttons don't work with French locale

mickael9, 2018-05-09:
This is because clicking the Upload button runs the task `PlatformIO: Upload`
https://github.com/platformio/platformio-vscode-ide/blob/develop/src/main.js#L181

However, with the French locale, a space is appended before the colon so the task name becomes `PlatformIO : Upload` and the command doesn't work (it prompts you to run any of the available tasks)

ivankravets, 2018-05-26:
Maybe that is a bug of locale? Why do they append this space?

mickael9, 2018-05-30:
I don't get why you think this is a locale bug; using a localisable string as an identifier is just wrong.

Tasks can be matched using via identifier as seen here : https://github.com/Microsoft/vscode/blob/master/src/vs/workbench/parts/tasks/electron-browser/task.contribution.ts#L1174. Why not use that instead?

ivankravets, 2018-05-30:
@dbaeumer Why does FRA locale use it as `PlatformIO : Upload`? Who does add this `space`?

ivankravets, 2018-05-30:
@mickael9  Wow! 😮 Thanks! Let's wait when @dbaeumer resolve https://github.com/Microsoft/vscode/issues/50799 . Then I'll update this extension.

dbaeumer, 2018-05-31:
@ivankravets alternatively you could use `executeTask` directly instead of going through the backdoor of executeCommand.

ivankravets, 2018-05-31:
Thanks, I thought the same yesterday. We need a new release of VSCode firstly.

dbaeumer, 2018-05-31:
Agree. Planed for beginning of next week.

closed 2018-08-28 by ivankravets

Navis-Raven, 2018-08-31:
It seems that it have been solved.

commit 01de86d6 (2018-08-28): Support for multi-root workspaces; Implement Project Indexer and Task Manager // Resolve #50, Resolve #107