platformio/platformio-vscode-ide #109: Selecting the target for build?

RudolphRiedel, 2018-05-12:
I am trying to compile Marlin 2.0 with PlatformIO and I can not figure out how to change the target.
Marlin comes with a platformio.ini file that defines several targets by platform, framework, board and so on.
I need to build for [env:sanguino_atmega1284p] but can not figure out how to select the target.

There is no config-button, no menu entry, no command-line to setup the env: to build with, right-clicking on build does nothing usefull.

How do I select the target from the supplied list?

ivankravets, 2018-05-12:
http://docs.platformio.org/en/latest/ide/vscode.html#task-runner

closed 2018-05-12 by ivankravets

RudolphRiedel, 2018-05-12:
Thank you for not really providing an answer to the question and just closing this instead.

ivankravets, 2018-05-12:
Have you read that doc? `VSCode: Menu > Tasks > Run Tasks...` and search for sanguino_atmega1284p

ivankravets, 2018-05-12:
We will add a separate PlatformIO Activity view to VSCode, you will be able to process build environment from the left side (tree)

RudolphRiedel, 2018-05-12:
I read the doc and could not figure out what it means as there is no Menu->Tasks in my vscode.
And now I just realised that is because my vscode has a German menu on top and what is listed under "Aufgaben" seems to be totally unrelated at first.
The whole menu looks like it belongs to vscode and not to PlatformIO.
I was looking for options in that menu on the bottom...

"By default, VS Code picks up the operating system's display language"...
I changed that to US now and it still is totally unclear that this is part of PlatformIO and I would never have found it even if my vscode had US as locale in the first place.

Sorry, not meaning to be rude, I just learned of PlatformIO today and think it is great!

But I could really not figure this out and localised menus are bad a idea.
In hope that my fail helps others...