platformio/platformio-vscode-ide #116: Allow to setup custom commands for buttons in PIO Toolbar

ivankravets, 2018-06-08:
Would be good to allow people to set up the custom commands for the buttons in PIO Toolbar.
For example, "Build" button now calls "PlatformIO: Build" task. We can create own task based on "PlatformIO: Build" and extended it with own arguments. Later, override "Build" button to "My custom task"

Niek, 2018-12-10:
This would be very useful. Right now it's impossible to specify an environment in the Build task. It's possible to create a custom task (``"type": "shell"``) and pass the environment in the ``-e`` argument, but this this not available for tasks with ``"type": "PlatformIO"``.

ivankravets, 2018-12-12:
Have you tried Left Sidebar > PlatformIO > Project Tasks > Expand your env > Build?

Niek, 2018-12-12:
Sure, that's possible. But it's not possible to change the default Build task in the bottom bar.

ivankravets, 2018-12-12:
See https://docs.platformio.org/en/latest/ide/vscode.html#custom-build-task

closed 2018-12-12 by ivankravets

ivankravets, 2018-12-12:
Please update the extension to 1.6.0 => https://marketplace.visualstudio.com/items/platformio.platformio-ide/changelog

commit 947bc1a4 (2018-12-12): Custom task for "Build" command which is used by PlatformIO Toolbar and Key Bindings // Resolve #116