platformio/platformio-vscode-ide #120: Start PIO Home Server

Sannioy, 2018-06-21:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.16.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Could not start PIO Home server: Error: timeout
```

ivankravets, 2018-06-21:
Could you reproduce this issue after a restart of VSCode?

ivankravets, 2018-06-25:
Could you reproduce this issue after a restart of VSCode?

/cc @rjbhide @clock199 @Sannioy

alexodus, 2018-06-25:
```
Traceback (most recent call last):
  File "main.py", line 17, in init main
  File "cmd_home.py", line 20, in init cmd_home
ImportError: No module named core
```

rjbhide, 2018-06-26:
had to install following missing dependencies
pip install constantly
pip install bs4
Then issue got resolved

ivankravets, 2018-06-26:
@rjbhide what is your OS? You don't need to install those packages manually.

ivankravets, 2018-06-26:
Fixed in the latest release. Please upgrade extension to the latest version, open VSCode terminal and type `pio upgrade --dev` and `pio update`.

closed 2018-06-26 by ivankravets

lmnewey, 2018-06-30:
It isn't fixed I've just followed your instructions and its still happening. I'm using both visual code and atom in windows 10

ivankravets, 2018-07-01:
Could you provide an error report?