platformio/platformio-vscode-ide #124: IntelliSense include errors

golyalpha, 2018-06-22:
IntelliSense can't seem to find avr/* libraries used by, among other things, Arduino.h

Results in IntelliSense marking it as a broken import (if using semantic IntelliSense).
Ends up looking something like this:
[image]

ivankravets, 2018-06-23:
Does it work after restart of VSCode?

golyalpha, 2018-06-23:
The same issue even after restart.

ivankravets, 2018-06-26:
Please attach here `.vscode/c_cpp_properties.json` file from this project.

ivankravets, 2018-08-03:
Please reopen if you still need help

closed 2018-08-03 by ivankravets

mutthunaveen, 2019-04-02:
Hello i ve the same problem here.. lot of retries but of no use :(

here is the file you requested
`{
    "configurations": [
        {
            "name": "!!! WARNING !!! AUTO-GENERATED FILE, PLEASE DO NOT MODIFY IT AND USE https://docs.platformio.org/page/projectconf/section_env_build.html#build-flags",
            "intelliSenseMode": "msvc-x64"
        },
        {
            "name": "Win32",
            "includePath": [
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/cores/arduino",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/variants/eightanaloginputs",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/EEPROM/src",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/HID/src",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SPI/src",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial/src",
                "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src",
                "C:/Users/Venu's/.platformio/packages/tool-unity",
                "c:/Users/Venu's/Documents/PlatformIO/Projects/blink/include",
                "c:/Users/Venu's/Documents/PlatformIO/Projects/blink/src",
                ""
            ],
            "browse": {
                "limitSymbolsToIncludedHeaders": true,
                "databaseFilename": "${workspaceRoot}/.vscode/.browse.c_cpp.db",
                "path": [
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/cores/arduino",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/variants/eightanaloginputs",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/EEPROM/src",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/HID/src",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SPI/src",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/SoftwareSerial/src",
                    "C:/Users/Venu's/.platformio/packages/framework-arduinoavr/libraries/__cores__/arduino/Wire/src",
                    "C:/Users/Venu's/.platformio/packages/tool-unity",
                    "c:/Users/Venu's/Documents/PlatformIO/Projects/blink/include",
                    "c:/Users/Venu's/Documents/PlatformIO/Projects/blink/src",
                    ""
                ]
            },
            "defines": [
                "PLATFORMIO=30606",
                "ARDUINO_AVR_NANO",
                "F_CPU=16000000L",
                "ARDUINO_ARCH_AVR",
                "ARDUINO=10805",
                "__AVR_ATmega328P__",
                ""
            ],
            "intelliSenseMode": "clang-x64",
            "cStandard": "c11",
            "cppStandard": "c++11",
            "compilerPath": "C:/Users/Venu's/.platformio/packages/toolchain-atmelavr/bin/avr-gcc.exe -mmcu=atmega328p"
        }
    ],
    "version": 4
}`

could you please resolve my issue. 

the iteratations that i ve applied are
1. adding stdlib.h path to the above file
2. untill compilation it works fine. after compiling my edited path will reset. (means the path ive added will get deleted)

ivankravets, 2019-04-02:
@mutthunaveen which path is missed?

mutthunaveen, 2019-04-03:
Hello Ivan, i ve opend new issue on the same here
[https://github.com/platformio/platformio-vscode-ide/issues/669](url)