platformio/platformio-vscode-ide #125: Installation Manager

vartasium, 2018-06-23:
# Description of problem
Failed to install PIO extension in both ATOM and VSCode with the same error

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.16.1
System: Windows_NT, 10.0.17134, x64

# Exception
```
Virtualenv Create: Traceback (most recent call last):
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 703, in <module>
    main()
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 692, in main
    aliasmbcs()
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 515, in aliasmbcs
    import locale, codecs
  File "C:\Users\root\.platformio\penv\lib\python2.7/locale.py", line 19, in <module>
    import functools
ImportError: No module named functools

```

ivankravets, 2018-06-25:
We have already fixed this issue in Atom extension. Could you re-test Atom? 

1) Please update PIO IDE for Atom via Atom > Settings > Updates to v2.1.8
2) Please remove `C:/Users/root/.platformio/penv` and `C:/Users/root/.platformio/python27` folders
3) Restart Atom.

vartasium, 2018-06-25:
no Updates to v2.1.8 available.
I reinstalled Atom and PIO IDE and this time got different error msg:

# Configuration
Atom: 1.28.0
PIO IDE: v2.1.7
System: Windows_NT, 10.0.17134, x64

# Exception
```
Install Virtualenv globally: C:/Qt/Tools/mingw530_32/opt/bin/python.exe: No module named pip

```

ivankravets, 2018-06-25:
Sorry, yes, v1.2.8. 
Please provide an output from system Terminal `cmd`
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

vartasium, 2018-06-25:
C:\Users\root>echo %PATH%
C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Libraries\redist\intel64\compiler;C:\ProgramData\Oracle\Java\javapath;C:\Program Files (x86)\Common Files\Intel\Shared Files\cpp\bin\Intel64;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Rhozet\RMP\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin\;C:\Program Files\IVI Foundation\VISA\Win64\Bin\;C:\Program Files (x86)\IVI Foundation\VISA\WinNT\Bin;C:\Program Files (x86)\QuickTime\QTSystem\;C:\Program Files (x86)\curl;;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Program Files\LLVM\bin;C:\Users\root\AppData\Local\Microsoft\WindowsApps;C:\Qt\Tools\mingw530_32\opt\bin;C:\Users\root\AppData\Roaming\npm;C:\Program Files\Microsoft VS Code\bin

C:\Users\root>echo %PYTHONPATH%
%PYTHONPATH%

C:\Users\root>python --version
Python 2.7.9

C:\Users\root>pip --version
'pip' is not recognized as an internal or external command,
operable program or batch file.

C:\Users\root>pip list
'pip' is not recognized as an internal or external command,
operable program or batch file.

ivankravets, 2018-06-25:
Thanks!

- Please close Atom and remove `C:/Users/root/.platformio/penv`
- Open Atom and toggle Menu > View > Developer > Toggle Developer tools

Please provide all errors from Console.

P.S: I see that you added very old Python to env PATH from Qt. Would be good to check Console logs what is going wrong.

vartasium, 2018-06-25:
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand platformio Array(1) Object
manager.js:89 'platformio' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn platformio ENOENT
(anonymous) @ manager.js:89
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand conda Array(1) Object
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand C:\Qt\Tools\mingw530_32\opt\bin\python.exe Array(2) Object
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand C:\Qt\Tools\mingw530_32\opt\bin\python.exe Array(5) Object
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand virtualenv Array(3) Object
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 User's Virtualenv: 'virtualenv' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn virtualenv ENOENT
(anonymous) @ C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand C:\Qt\Tools\mingw530_32\opt\bin\python.exe Array(2) Object
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 Virtualenv Create: Traceback (most recent call last):
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 703, in <module>
    main()
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 692, in main
    aliasmbcs()
  File "C:/Users/root/.platformio/penv/lib/python2.7/site.py", line 515, in aliasmbcs
    import locale, codecs
  File "C:\Users\root\.platformio\penv\lib\python2.7/locale.py", line 19, in <module>
    import functools
ImportError: No module named functools

(anonymous) @ C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1
C:\Users\root\.atom\packages\platformio-ide\node_modules\platformio-node-helpers\lib\index.js:1 runCommand C:\Qt\Tools\mingw530_32\opt\bin\python.exe Array(4) Object
utils.js:50 InstallationManager Install Virtualenv globally: C:/Qt/Tools/mingw530_32/opt/bin/python.exe: No module named pip

notifyError @ utils.js:50

ivankravets, 2018-06-25:
Does it work when you run directly from Windows `cmd`?
```
python -c "import functools; print functools"
```

vartasium, 2018-06-25:
C:\Users\root>python -c "import functools; print functools"
<module 'functools' from 'C:\Qt\Tools\mingw530_32\opt\lib\python2.7/functools.pyc'>

ivankravets, 2018-06-26:
Fixed in v0.16.2. Please update and restart VSCode.

closed 2018-06-26 by ivankravets

vartasium, 2018-06-26:
Installation was successful.
Thank you!