platformio/platformio-vscode-ide #128: Longer builds fail within VS Code

robbertkl, 2018-06-26:
Having used PlatformIO before, I decided to give the VS Code IDE a try. While it worked quite well for small projects, when I tried to compile a larger project ([Marlin](https://github.com/MarlinFirmware/Marlin) in this case) it fails every time, but always at a different place / module. The error is always something like "file not found" or "no such file or directory".

I've tried running PlatformIO in my terminal, with the same PlatformIO Core that was installed by the IDE (by adding `~/.platformio/penv/bin` to my `$PATH`), which runs fine.

Then I tried running PlatformIO in my terminal again, but this time with the VS Code project open at the same time. This time it failed in my terminal as well!

It seems some background process within VSCode / PlatformIO IDE is doing some automated cleanup of the `.pioenvs` directory mid-compile, removing build products needed in later steps and therefore failing the entire build.

I run macOS 10.13.5, VS Code 1.24.1, PlatformIO IDE 0.16.1 which has installed PlatformIO Core 3.6.0a3.

ivankravets, 2018-06-26:
Could you provide a screenshot of VSCode IDE terminal with these errors?

robbertkl, 2018-06-26:
Here's a screenshot:

[image]

But like I said, the exact error message and module where it fails is different each time.

robbertkl, 2018-06-26:
I think I've managed to find the culprit: setting `platformio.autoUpdateIncludes` to `false` makes the compile complete successfully.

robbertkl, 2018-06-26:
Just found out `platformio.autoUpdateIncludes` is a setting from the [IoT Utility plugin](https://marketplace.visualstudio.com/items?itemName=formulahendry.platformio) and not the PlatformIO IDE. I guess I somehow had this plugin already on my system, and it was conflicting with PlatformIO IDE.

That should explain it. Removed the IoT Utility plugin.

closed 2018-06-26 by robbertkl