platformio/platformio-vscode-ide #131: Start PIO Home Server

rsilvas, 2018-06-27:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.16.2
System: Windows_NT, 10.0.17134, x64

# Exception
```
Please manually remove file `C:\Users\sunus\.platformio\packages\tool-pioplus\pioplus.exe`
Please manually remove file `C:\Users\sunus\.platformio\packages\tool-pioplus\pioplus.exe`
Error: Could not install 'tool-pioplus' with version requirements '>=1.3.1,<2' for your system 'windows_amd64'.

Please try this solution -> http://bit.ly/faq-package-manager

```

ivankravets, 2018-07-07:
@StephanJoubert did this solution help?

ivankravets, 2018-08-03:
Fixed in the latest release.

closed 2018-08-03 by ivankravets