platformio/platformio-vscode-ide #140: Start PIO Home Server

pankleks, 2018-06-30:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.17.0
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\users\root\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\users\root\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\users\root\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\users\root\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\Python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

```

pankleks, 2018-07-01:
I reinstalled python 2.7.14 to 2.7.15 and uninstalled VS extensions and installed it again at it helped.

Not sure it's related to new python version or more to fact that installation possibly fixed paths etc.

ivankravets, 2018-07-07:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/142

This is a known issue :( I can't reproduce it but I saw it in action. The same, removing Python and installing again fixes that.

closed 2018-07-07 by ivankravets