platformio/platformio-vscode-ide #151: Installation Manager

stanokudlac, 2018-07-06:
# Description of problem
Failed to install PlatformIO IDE for VS Code

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.17.1
System: Linux, 4.4.0-130-generic, x64

# Exception
```
Error: Could not create PIO Core Virtual Environment. Please create it manually -> http://bit.ly/pio-core-virtualenv
	at t.<anonymous> (/home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:21112)
	at P (/home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:33993)
	at Generator._invoke (/home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:33781)
	at Generator.e.(anonymous function) [as throw] (/home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:34172)
	at n (/home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:13752)
	at /home/stando/.vscode/extensions/platformio.platformio-ide-0.17.1/node_modules/platformio-node-helpers/lib/index.js:1:13871
	at <anonymous>
```

ivankravets, 2018-07-06:
Have you tried http://bit.ly/pio-core-virtualenv ?

stanokudlac, 2018-07-06:
Yeah, meanwhile I have found this solution as well. And it works! Thanks a
lot for you support.

ivankravets, 2018-07-06:
What is your Linux distribution?

stanokudlac, 2018-07-06:
It's 4.4.0-130-generic #156~14.04.1-Ubuntu SMP x86_64 GNU/Linux

ivankravets, 2018-07-07:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/154

ivankravets, 2018-07-07:
Nevertheless, please provide an output from a system Terminal
```
echo $PATH
echo $PYTHONPATH
python --version
pip --version
pip list
```

closed 2018-07-07 by ivankravets

ivankravets, 2018-07-09:
Your environment looks well. Could you try to remove `~/.platformio/penv` folder and restart VSCode?
If you see errors again, please provide Console Output from VSCode > Help > Toggle Developer Tools > Console. Thanks!

stanokudlac, 2018-07-13:
Here it is

mirao, 2019-04-22:
I got the same error during installation (Failed to install PlatformIO IDE), however I didn't see any message like "Error: Could not create PIO Core Virtual Environment".
As @ivankravets suggested, I removed `~/.platformio/penv` and restarted VSCode. Now installation has finished successfully:
```
Installing PlatformIO Core...
Please do not close this window and do not open other folders until this process is completed.
PlatformIO IDE installed successfully.

Please restart VSCode.
```
One more VSCode's reload and I got the PIO homepage.

**Used SW:**
- Ubuntu 19.04
- VSCode 1.33.1
- PlatformIO IDE 1.7.1
- Python 2.7.16
- pip 18.1