platformio/platformio-vscode-ide #155: Start PIO Home Server

ofiratt, 2018-07-07:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.17.1
System: Windows_NT, 10.0.16299, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\ofira\.platformio\packages\contrib-pysite\twisted\internet\protocol.py", line 17, in <module>
    from twisted.python import log, failure, components
  File "C:\Users\ofira\.platformio\packages\contrib-pysite\twisted\python\log.py", line 26, in <module>
    from twisted.logger import (
  File "C:\Users\ofira\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 89, in <module>
    from ._levels import InvalidLogLevelError, LogLevel
  File "C:\Users\ofira\.platformio\packages\contrib-pysite\twisted\logger\_levels.py", line 9, in <module>
    from constantly import NamedConstant, Names
ImportError: No module named constantly

```

ivankravets, 2018-07-07:
Please remove `C:\Users\ofira\.platformio\packages\contrib-pysite` folder and restart VSCode. Did it help?

Sambo007, 2018-07-11:
Today I had the same issue. So based on Ivan's suggestion, I renamed the 'contrib-pysite' folder by adding 'contrib-pysite.del' and it did not help. However, when I moved it out of the packages folder the error went away! I was afraid to delete it at first.

The deleted folder got recreated automatically after PIO home screen showed up.

BTW, this also helped resolve an issue I had with PIO not starting automatically in VS Code. I always had to start it manually from the 'Command Palette...' I searched thoroughly for a solution to this problem but to no avail! Now I got 2 birds in 1 stone! :)

You're the man Ivan! Thanks a lot!

Sam

ivankravets, 2018-07-12:
Fixed in https://github.com/platformio/platformio-core/commit/44a926b30a1b17b31793b05844aa64c8223708db

Please remove `C:\Users\ofira\.platformio\packages` folder 
and open PIO IDE Terminal and type:
```
pio upgrade --dev
pio update
```

Restart VSCode. Does it work now?

closed 2018-07-12 by ivankravets

stritti, 2018-07-17:
In addition I had to kill the task `pioplus.exe` on Windows 10. Then @ivankravets suggestion worked for me.