platformio/platformio-vscode-ide #157: Start PIO Home Server

ssqf, 2018-07-09:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.0
PIO IDE: v0.17.1
System: Windows_NT, 10.0.16299, x64

# Exception
```
Error: Traceback (most recent call last):
  File "c:\users\administrator\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\administrator\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\administrator\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\administrator\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\administrator\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\administrator\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\administrator\.platformio\penv\lib\site-packages\platformio\commands\home.py", line 32, in cli
    pioplus_call(sys.argv[1:])
  File "c:\users\administrator\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 114, in pioplus_call
    code = subprocess.call([pioplus_path] + args, **kwargs)
  File "C:\Users\Administrator\.platformio\python27\Lib\subprocess.py", line 172, in call
    return Popen(*popenargs, **kwargs).wait()
  File "C:\Users\Administrator\.platformio\python27\Lib\subprocess.py", line 394, in __init__
    errread, errwrite)
  File "C:\Users\Administrator\.platformio\python27\Lib\subprocess.py", line 644, in _execute_child
    startupinfo)
WindowsError: [Error 2] 

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

ivankravets, 2018-07-10:
Could you reproduce this issue after restart?

ivankravets, 2018-08-03:
Please reopen if you still need help

MkLHX, 2018-08-07:
Hi!
I have the same error.
The issue is reproduce allways i start VScode. I'have this message :  
[image]

ivankravets, 2018-08-07:
It seems that antivirus tool blocks/removes PIO Plus Core from `c:\users\YourUserName\.platformio\packages\tool-pioplus`. Please check if you have this folder or remove it manually and restart VSCode.

Please report if it helped.

MkLHX, 2018-08-07:
I disabled defender and add the folder in defender's exclusions : 
```
c:\users\YourUserName\.platformio\packages\tool-pioplus
```

So i get this issue now:  
```

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.4
System: Windows_NT, 10.0.17134, x64

# Exception
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\drumt\.platformio\packages\contrib-pysite\twisted\internet\protocol.py", line 15, in <module>
    from zope.interface import implementer
ImportError: No module named zope.interface

```

MkLHX, 2018-08-07:
Ok so i excluded c:\user\<MyUsername>\.platformio and i'll still trying use platformio ;-)

ivankravets, 2018-08-07:
I'm not sure that all files are valid in `C:\Users\drumt\.platformio`. I recommend to close VSCode, navigate to `C:\Users\drumt\.platformio`, remove all files and folders. Now, start VSCode.

brogamma, 2018-09-11:
Hi, I tried to resolve this issue but even adding the folder `C:\Users\drumt\.platformio` to the exlusion folder of defender i still get the same error code. Even after following the steps shown above.

MkLHX, 2018-09-11:
Hello, you need to use your username.

c:\users\sven\.platformio

brogamma, 2018-09-11:
I used the username in the way you provided but the error keeps existing. I have no clue where the issue may come from.

MkLHX, 2018-09-11:
Have you removed the .platformio folder ? And retry ?

brogamma, 2018-09-11:
I tried removing the folder and restarting. Reinstalling the extension and i added the .platfromIO forlder to the exceptions list of windows defender. Nothing helped.

ivankravets, 2018-09-11:
@brogamma 

1. Please open Start > Control Panel > Installed Programs > Remove ALL Pythons installed in a system
2. Remove `C:\Users\<user>\.platformio` folder
3. Start VSCode, check that you use the latest v1.1.1 version of our extension.

brogamma, 2018-09-11:
@ivankravets That helped! Had Python 2.7.1 installed and uninstallig that made the extension work. Thanks for the help!

ivankravets, 2018-10-26:
Resolved in the latest release.

closed 2018-10-26 by ivankravets