platformio/platformio-vscode-ide #160: Start PIO Home Server

ittchmh, 2018-07-11:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.1
PIO IDE: v0.17.2
System: Windows_NT, 10.0.14393, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\hmh\.platformio\packages\contrib-pysite\twisted\internet\protocol.py", line 17, in <module>
    from twisted.python import log, failure, components
  File "C:\Users\hmh\.platformio\packages\contrib-pysite\twisted\python\log.py", line 26, in <module>
    from twisted.logger import (
  File "C:\Users\hmh\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 89, in <module>
    from ._levels import InvalidLogLevelError, LogLevel
  File "C:\Users\hmh\.platformio\packages\contrib-pysite\twisted\logger\_levels.py", line 9, in <module>
    from constantly import NamedConstant, Names
ImportError: No module named constantly

```

ittchmh, 2018-07-11:
I got issue after module update
Delete profile folder .platformio resolved issue

ivankravets, 2018-07-12:
Do you have antivirus system?

closed 2018-07-12 by ivankravets

ittchmh, 2018-07-12:
No, I don't have antivirus.