platformio/platformio-vscode-ide #165: Start PIO Home Server

arunsoman, 2018-07-13:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.0
PIO IDE: v0.17.2
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\Arun Soman\.platformio\packages\contrib-pysite\twisted\internet\protocol.py", line 17, in <module>
    from twisted.python import log, failure, components
  File "C:\Users\Arun Soman\.platformio\packages\contrib-pysite\twisted\python\log.py", line 26, in <module>
    from twisted.logger import (
  File "C:\Users\Arun Soman\.platformio\packages\contrib-pysite\twisted\logger\__init__.py", line 89, in <module>
    from ._levels import InvalidLogLevelError, LogLevel
  File "C:\Users\Arun Soman\.platformio\packages\contrib-pysite\twisted\logger\_levels.py", line 9, in <module>
    from constantly import NamedConstant, Names
ImportError: No module named constantly

```

ivankravets, 2018-07-13:
Could you archive this folder `C:\Users\Arun Soman\.platformio\packages\contrib-pysite` and share on Dropbox or another service?

ivankravets, 2018-07-17:
1. Please remove this folder `%USER_HOME%\.platformio\packages\contrib-pysite` 
2. Open PlatformIO IDE
3. Open PlatformIO IDE Terminal and type
```
pio upgrade --dev
pio update
```

Restart IDE. Everything should work now.

ivankravets, 2018-08-03:
Should be fixed in the latest release

closed 2018-08-03 by ivankravets