platformio/platformio-vscode-ide #169: I have an established Git client Git-2.18.0-64

tia777, 2018-07-13:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.2
System: Windows_NT, 6.1.7601, x64

# Exception
```
Error: Please install Git client from https://git-scm.com/downloads

```

ivankravets, 2018-07-17:
It should be added to system PATH

closed 2018-07-17 by ivankravets

tia777, 2018-07-19:
куда надо отправить этот файл ? можно точнее ? и хочу узнать как добавить библиотеки от Arduino в  программу ? Еще не могу зайти в домашнюю страницу Pio Home

ivankravets, 2018-07-19:
> и хочу узнать как добавить библиотеки от Arduino в программу ?

* http://docs.platformio.org/en/latest/projectconf/section_env_library.html#lib-deps

> Еще не могу зайти в домашнюю страницу Pio Home

Could you provide more details? Do you see any errors?

ivankravets, 2018-07-19:
Menu: Help > Toggle Developer Tools > Console. Do you see any errors?

Please try to open PIO IDE Terminal (see icon in bottom toolbar) and type `pio home`. Do you see PIO Home in browser?

tia777, 2018-07-19:
​вот все ошибки . pio home значек я вижу

ivankravets, 2018-07-19:
> Please try to open PIO IDE Terminal (see icon in bottom toolbar) and type pio home
```
pio home
```