platformio/platformio-vscode-ide #176: Start PIO Home Server

KevinDonnelly, 2018-07-17:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.2
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\me\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\me\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\me\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\me\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 7, in <module>
    from socket import error as SocketError, timeout as SocketTimeout
  File "c:\users\me\.platformio\penv\lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: The specified procedure could not be found.

```

KevinDonnelly, 2018-07-17:
HP support assistant blocking this software

ivankravets, 2018-07-18:
Could you allow 127.0.0.1:8010 for this app?

ivankravets, 2019-05-30:
Should be fixed in the latest release.

closed 2019-05-30 by ivankravets