platformio/platformio-vscode-ide #183: Start PIO Home Server

mshahrasbi, 2018-07-18:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.2
System: Windows_NT, 10.0.16299, ia32

# Exception
```
Error: Traceback (most recent call last):
  File "main.py", line 130, in main.main
  File "c:\users\mohammad.shahrasbi\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\mohammad.shahrasbi\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\mohammad.shahrasbi\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\mohammad.shahrasbi\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\mohammad.shahrasbi\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "cmd_home.py", line 806, in cmd_home.cli
  File "C:\Users\mohammad.shahrasbi\.platformio\packages\contrib-pysite\twisted\internet\posixbase.py", line 495, in listenTCP
    p.startListening()
  File "C:\Users\mohammad.shahrasbi\.platformio\packages\contrib-pysite\twisted\internet\tcp.py", line 1363, in startListening
    raise CannotListenError(self.interface, self.port, le)
CannotListenError: Couldn't listen on 127.0.0.1:8010: [Errno 10048] Only one usage of each socket address (protocol/network address/port) is normally permitted.

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to support@pioplus.com

============================================================

```

ivankravets, 2018-07-18:
Could you reproduce this issue after IDE restart?

mshahrasbi, 2018-07-18:
No, after restart IDE this is not happening...

ivankravets, 2018-07-18:
Thanks! Please reopen this issue if you will reproduce this issue again.

closed 2018-07-18 by ivankravets