platformio/platformio-vscode-ide #188: Start PIO Home Server

djewiss, 2018-07-20:
# Description of problem
Leave a comment...

Installed python 2.7.15, now platformio home does not start without an error. Build, upload etc still works OK.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 16, in init main
  File "cmd_debug.py", line 24, in init cmd_debug
  File "C:\Users\darryl\.platformio\packages\contrib-pysite\twisted\internet\stdio.py", line 30, in <module>
    from twisted.internet import _win32stdio
  File "C:\Users\darryl\.platformio\packages\contrib-pysite\twisted\internet\_win32stdio.py", line 9, in <module>
    import win32api
ImportError: DLL load failed: %1 is not a valid Win32 application.

```

ivankravets, 2018-07-20:
1. Try to uninstall all Pythons from a system
2. Remove this folder `C:\Users\darryl\.platformio\penv`
3. Install Python 2.7.15
4. Start IDE

djewiss, 2018-07-21:
That worked, it's all good now, thanks!

closed 2018-07-21 by djewiss