platformio/platformio-vscode-ide #192: Start PIO Home Server

Costas-Zeimpekis, 2018-07-20:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.17134, x64

# Exception
```
Error: Traceback (most recent call last):
  File "c:\users\costas\.platformio\penv\lib\site-packages\platformio\__main__.py", line 120, in main
    cli(None, None, None)
  File "c:\users\costas\.platformio\penv\lib\site-packages\click\core.py", line 700, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\costas\.platformio\penv\lib\site-packages\click\core.py", line 680, in main
    rv = self.invoke(ctx)
  File "c:\users\costas\.platformio\penv\lib\site-packages\click\core.py", line 1027, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "c:\users\costas\.platformio\penv\lib\site-packages\click\core.py", line 873, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\costas\.platformio\penv\lib\site-packages\click\core.py", line 508, in invoke
    return callback(*args, **kwargs)
  File "c:\users\costas\.platformio\penv\lib\site-packages\platformio\commands\home.py", line 32, in cli
    pioplus_call(sys.argv[1:])
  File "c:\users\costas\.platformio\penv\lib\site-packages\platformio\managers\core.py", line 114, in pioplus_call
    code = subprocess.call([pioplus_path] + args, **kwargs)
  File "C:\Python27\Lib\subprocess.py", line 168, in call
    return Popen(*popenargs, **kwargs).wait()
  File "C:\Python27\Lib\subprocess.py", line 390, in __init__
    errread, errwrite)
  File "C:\Python27\Lib\subprocess.py", line 640, in _execute_child
    startupinfo)
WindowsError: [Error 5] Access is denied

============================================================

An unexpected error occurred. Further steps:

* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command

* Try to find answer in FAQ Troubleshooting section
  http://docs.platformio.org/page/faq.html

* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues

============================================================

```

ivankravets, 2018-07-24:
Do you have antivirus tools in a system? Could you try to disable them for a while?

laurentopia, 2018-08-12:
Getting this error when uploading, it requires restarting vscode to regain the ability to upload
serial.serialutil.SerialException: could not open port 'COM5': WindowsError(5, 'Access is denied.')
https://github.com/platformio/platformio-vscode-ide/issues/263

jpoles1, 2018-08-26:
Having this same issue, but a restart only fixes it occasionally

closed 2018-11-18 by ivankravets