platformio/platformio-vscode-ide #193: Start PIO Home Server

KevinDonnelly, 2018-07-20:
# Description of problem
Leave a comment...Your dumb jackass program is looking for a file on a USB which has nothing to do with the program. It is looking for powershell almost fanatically in a network location which has noting to do with studio code. The USB stick is used for my legal and financial documents. What are you playing at? Writing a program to access someone's personal information without their consent is not on buddy :A system error occurred (ENOENT: no such file or directory, mkdir 'd:\000WIBOWS')
BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\me\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\me\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\me\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\me\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 7, in <module>
    from socket import error as SocketError, timeout as SocketTimeout
  File "c:\users\me\.platformio\penv\lib\socket.py", line 47, in <module>
    import _socket
ImportError: DLL load failed: The specified procedure could not be found.

```

KevinDonnelly, 2018-07-26:
Correct me if I am wrong.

If the program 'Visual Studio Code fetches extensions: (function  X)

Then why is the  program Visual Studio Code fetching extensions from not(function X)

That is can you explain why it was trolling thru a removable drive?

ivankravets, 2018-07-26:
> That is can you explain why it was trolling thru a removable drive?

How did you check it? Could you show some logs/screenshots? To be honest,  I don't understand what do you mean :(