platformio/platformio-vscode-ide #202: Start PIO Home Server

Gecooo, 2018-07-23:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.17134, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 12, in init main
  File "c:\.platformio\penv\lib\site-packages\click\__init__.py", line 18, in <module>
    from .core import Context, BaseCommand, Command, MultiCommand, Group, \
  File "c:\.platformio\penv\lib\site-packages\click\core.py", line 8, in <module>
    from .types import convert_type, IntRange, BOOL
  File "c:\.platformio\penv\lib\site-packages\click\types.py", line 5, in <module>
    from ._compat import open_stream, text_type, filename_to_ui, \
  File "c:\.platformio\penv\lib\site-packages\click\_compat.py", line 6, in <module>
    from weakref import WeakKeyDictionary
  File "C:\.platformio\python27\Lib\weakref.py", line 14, in <module>
    from _weakref import (
ImportError: cannot import name _remove_dead_weakref

```

Gecooo, 2018-07-23:
Get a copy of python 2.7.0 and overwrite the contents of the platform io python27 dir. In your case: C:\Users\ikevin2810.platformio\python27\

its worked.

ivankravets, 2018-07-24:
Duplicate of https://github.com/platformio/platformio-vscode-ide/issues/142

closed 2018-07-24 by ivankravets