platformio/platformio-vscode-ide #203: oddness with colorization of inactive preprocessor blocks

sstaub, 2018-07-23:
macOS Mojave 10.14 Beta (18A336e)
VS Code 1.25.1 / 1.26-insider
C/C++ 0.17.7
PlatformIO IDE 0.17.3
used board Fubarino Mini with PIC32MX micro controller

That issue is about the nice feature to dim down unused preprocessor blocks
in the code.
In the moment there are some wrong preprocessor block marked as inactive.
This happens with micro controller specific definitions.

#if defined(PIC32MX)
#define TYP 2 // this part is marked as inactive, but it is wrong
#else
#define TYP 0 // this part is marked active, but it is also wrong
#endif

see screenshot
[image]

I made an issue on cpp-tools (#2304) and the solution is to add the macro in the c_cpp_properties.json file as define
[image]

The solution is to add automatic the right processor family (PIC32MX) and type (32MX250F128B) definitions to the c_cpp_properties.json file. Maybe other infos like f_cpu are also useful.

I opened also an issue on vscode-arduino (#615), there is the same problem.

ivankravets, 2018-07-24:
Please move this issue to 
https://github.com/platformio/platform-microchippic32/issue and highlight which macros need to specify.

sstaub, 2018-07-24:
I will have a look what happens with other microcontrollers, but I think that is generall problem.