platformio/platformio-vscode-ide #205: Start PIO Home Server

AmrKhaledNowahy, 2018-07-24:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.10240, x64

# Exception
```
Could not start PIO Home server: Error: timeout
```

ivankravets, 2018-08-30:
Please reopen if you still need help

figure0, 2019-04-30:
Could not start PIO Home server: Error: timeout #205

NitrofMtl, 2020-09-20:
after running pio home, restart IDE, pio loading page stay stuck on : 

     PlatformIO: IntelliSense Index Rebuild

WIN10
vscode 1.49.1
PIO 2.1.0

sebby49, 2020-10-18:
i got the same problem as NitrofMtl

sebby49, 2020-10-19:
[image]
can somebody tell me why its not working as it should i been messing with this for a week already reading all kinds of solutions but nothing works like wtf is it so hard to just put a firmware out there that works so noobs like me don't get seriously frustrated

ivankravets, 2020-10-19:
Do you use antivirus software?

sebby49, 2020-10-19:
Yes its AVG and i had turned it of before and that didnt workVerzonden vanaf mijn Samsung Galaxy-smartphone.

ivankravets, 2020-10-19:
It blocks PlatformIO from running PlatformIO home local server. 

You can try to run it from PlatformIO CLI
`pio home`

Does it work?

P.S: there is nothing useful from today’s antivirus software. It damages your machines and walks around your files looking for sweet things. Check Internet what Avast does in user machines.

sebby49, 2020-10-19:
i already noticed i can turn of the firewall but the AVG engine i can't turn of means i have to uninstal it to shut down also i am a complete noob in coding etc so to me its all gibberisch

sebby49, 2020-10-19:
now i uninstalled AVG software and this is what i get same thing
[image]

ivankravets, 2020-10-19:
Check a debug information, see details in the output window.

sebby49, 2020-10-19:
[image]
hmm

ivankravets, 2020-10-19:
Do you use the latest PlatformIO IDE extension 2.1.3? Please wait for a few seconds when PlatformIO loads, then start debug.

sebby49, 2020-10-19:
yes i have the latest version and it fails to install the IO core and now the intellisense index rebuild keeps spinning forever

ivankravets, 2020-10-19:
Sorry, I can't help here. It seems you broke VSCode configuration. Please open VSCode preferences for PlatformIO IDE and RESET evrything to the default state. Also, you can remove `%HOME_DIR%/.platformio` folder and restart VSCode.

I recommend to open discussin at https://community.platformio.org/. We don't provide help here.

haloyrap, 2021-05-16:
yo soluciones buscando la carpeta .platformio y pegándola en el disco local de mi sistema operativo. no es crear una carpeta es buscar la carpeta y pegarla.

cpvcristian, 2021-06-08:
> 
> 
> yo soluciones buscando la carpeta .platformio y pegándola en el disco local de mi sistema operativo. no es crear una carpeta es buscar la carpeta y pegarla.

Buenas, de dónde la copiaste y dónde la pegaste?

shj8893, 2021-06-12:
After I install WSL2, PlatformIO doesn't works.!
WSL2 uses 127.0.0.1 as it's local domain. And PIO also seems to use the same URL.

mick-git, 2021-07-23:
Ola!

Não consigo instala a plataformaIO, segue em anexo a falha ... Até desinstalei o python, mas segue dando falha.

[image]
 Por favor me ajuda!

luigimannoni, 2021-08-05:
I was having the same problems and have fixed this on Ubuntu 21.04:

- I had python3.9 installed but platformio was using python3.7 and trying to access some of the libraries from 3.7 which were not accessible.
- I have uninstalled the semantic-version using pip 3.9 `pip3.9 uninstall semantic-version`
- Python3.7 on ubuntu 21.04 is broken
- Installed pyenv
- `pyenv install 3.7.11`
- `pyenv global 3.7.11`
- Installed semantic-version and platformio `pip install semantic-version platformio`
- Restart VSCode

evex111, 2021-08-24:
Tenia el mismo problema y me funciono
* desinstalar visual y todas las carpetas relacionada 
* volvi a instalarlo después Python 
* En el buscador pondras "alias" y desactivas los instaladores de Python, volví a instalar platformIO y listo

lmindiola, 2021-09-18:
en mac os lo que me funciono 
- Ir a la extensión
- Configuración de extensión
- Platformio-ide: Custom PATH
- mi ruta fue `/Users/usuario/.platformio`

la carpeta de platformio no la movi, esa fue la ruta por defecto de la instalación.

JonathanCaes, 2021-10-03:
I had the same issue on macOS Big Sur Version 11.6. I resolved the issue by downgrading to a previous version of platformIO and then updating to the newest version again. I thought I'd share this, since it might help someone else. 🙂

ivankravets, 2021-10-09:
@JonathanCaes could you update PlatformIO IDE extension to the latest version, removed folders below, and restart VSCode?
- `~/.platformio/python3`
- `~/.platformio/penv`

It seems the issue is linked with the outdated Python 3.8 version. 

Does it work now?

Lissanro, 2021-11-03:
I had this issue, and nothing helped. To investigate, I installed with pip:

    sudo -H pip install platformio

Then I could run `strace pio home` command in the terminal and I noticed it hangs on accessing `/run/uuidd/request` - so I removed it:

    sudo rm /run/uuidd/request

Then `pio home` command  could finish successfully. I installed PIO in vscode again, and it finally worked!

migueldiago, 2021-11-17:
I "solved" the issue in Kubuntu 20.04 by downgrading from v2.4.0 to the previous version (v2.3.4). It looks like it's working now.

ivankravets, 2021-11-17:
@migueldiago what was your error message?

x2764tech, 2021-12-17:
(Win10) Windows Firewall didn't prompt me to add rules for platformio.exe - doing this manually solved the issue

bestboxhenry, 2022-02-13:
好像并没有解决

Franciraldo, 2022-04-07:
On Window 10, 11, It can be only network driver error, just restart the network service running as administrator:
net stop winnat
net start winnat.

jonathand03, 2022-05-01:
just install pip from python3 : sudo apt install python3-pip, worked for me! and reinstall the extesion again! 
pio upgrade gave me the suggestion to install pip

mhakimih1988, 2022-05-07:
Thank you dear @jonathand03 
That worked for me as well.

growolff, 2022-05-09:
> This worked for me
`pip3 install semantic-version platformio`

Then restarted vscode and everything is ok now.

terdong, 2022-05-26:
Change Pio Home Server Http Port number to any available port from 0 like this image below.

[image]

NiklasProbst, 2022-06-07:
I tried a really long time but now I got PlatformIO working again. A 100% fix is to delete .platformio and .vscode folder in C:\users\<User>\. ATTENTION: all your vscode extensions will be gone.

5t1ckyF4pD4p, 2022-06-19:
I uninstalled and reinstalled everything many times and "pip3 install semantic-version platformio" worked for me, too. Thank you growolff!!!!

somehibs, 2022-06-22:
Platform IO Home would lock up when opening VSCode, apparently during Intellisense Index rebuild.

This happened after I deleted my .platformio folder for the first time in a long time.
The root cause is actually re-downloading all dependencies. 

While dependencies are downloading, there is no output in VSCode. This causes a UX issue, where the user is waiting for a PlatformIO VSCode Task to complete, but the name of the task is unrelated to the actual task being performed by pio.

Waiting for all of the dependencies to install by watching the contents of .platformio/packages slowly fill up gave me enough patience to wait for the task to complete. When it did, my IDE returned back to normal.

Simply, don't bother deleting any other folder than .platformio, and make sure you wait patiently for your libraries to re-download. I suggest monitoring network traffic, or the output folder.

Rafa121212, 2022-06-25:
Sorry please. I´m sorry . Upload fail!!
Is solved close atom and reopen vscode .

Sorry for the incovenence

dvlopez2010, 2022-07-05:
I battled this issue for a hot minute. I solved it for windows. I had to uninstall vs code. I then had to delete all files and folder for vs code and platformio. I was then able to reinstall vs code and platformio individually. now it opens and runs...

closed 2022-07-06 by ivankravets

Dragondude1960, 2022-07-19:
I am having this problem on Ubuntu 22.04. Latest VScode and latest PIO. It works fine on Win10 but not on Linux.

ivankravets, 2022-07-19:
@Dragondude1960 does it work if you open [PlatformIO CLI](https://docs.platformio.org/en/latest/integration/ide/vscode.html#platformio-core-cli) and type `pio home`?

dangcongtao, 2022-08-16:
> I had this issue, and nothing helped. To investigate, I installed with pip:
> 
> ```
> sudo -H pip install platformio
> ```
> 
> Then I could run `strace pio home` command in the terminal and I noticed it hangs on accessing `/run/uuidd/request` - so I removed it:
> 
> ```
> sudo rm /run/uuidd/request
> ```
> 
> Then after forcefully killing the process, when I ran `pio home` command again, it could finish successfully. I installed PIO in vscode again, and it finally worked!

Yess. thanks for that it helps me solve the same problem(Ubuntu 22.04 LTS with Vs Code)

lingdeyiyicifang, 2022-08-16:
it may due to the UI language, i changed the language to English and the problem solved.

Erriez, 2022-08-17:
I had the same issue on Ubuntu 22.04 and Visual Code to v1.70.1 (currently the latest). Resolved after `sudo apt install platformio` and running command `pio home` in shell (`Visual Code | PlatformIO Core CLI`) which starts the web browser correctly. Then restart PlatformIO. Thanks!

tiberilucaSPI, 2022-08-26:
i had the same issue on windows 7, visual studio code 1.70.2, i tryed every solution but nothing.

- uninstall, deleting directory .vscode and .platformio, reinstall -> nothing
- uninstall, deleting directory .vscode and .platformio, reinstall in different location -> nothing
- disabled every security program (windows firewall, defender)
- changed ui language

A strange fact is that the server always choose the 8008 port, even if i change it manually
even if i change the pio home server http port fro

For knowledge, my pc is under a corporate network that use a self signed certificate but i don't think is it because it work well on my laptop.

thanks

obiscr, 2022-08-31:
I fixed it by **disconnected networks**.

**Step 1**
Just disconnected your computer network, then execute `pio home`, obviously you will get the error. 

**Step 2**
Then reconnected your network, execute `pio home` again, it's working. 

I documented this issue on [my blog](https://www.obiscr.com/article/Could-not-start-PIO-Home-server).

It was confusing, but this action did solve my problem.

AlonsoSolis, 2022-09-23:
Hello I got this error since I install the Platformio:
[image]

vladimirdjuricic, 2022-10-07:
`pip install platformio`

before that upgrade pip, if needed:
`/usr/bin/python3 -m pip install --upgrade pip`

Tranks27, 2022-10-22:
My problem was with python installation. I had a fresh ubuntu 20.04 image, so I need to install some python dev tools.
`sudo apt update`
`sudo apt -y upgrade`
`python3 -V`
`sudo apt install -y build-essential libssl-dev libffi-dev python3-dev`

colorker, 2022-10-25:
> 将 Pio Home Server Http 端口号从 0 更改为任何可用端口，如下图所示。
> 
> [image]

think you! It goog use for me.

vky2207, 2022-10-31:
> Hello I got this error since I install the Platformio: [image]

me too eror in time out

vky2207, 2022-10-31:
[image]
hallo guys can help me eror this one?

vky2207, 2022-10-31:
[image]
please help eror "was not declared in this scope

rocketfuze, 2022-11-03:
I have had this problem for months as well with my Windows 10 PC on a corporate network.  I read through all of the above comments several times, trying different solutions including:  VSCode and platformio reinstalls (including running VSCode in portable mode), .platformio/.vscode folder/subfolder deletions, using different home server ports, attempting to whitelist ports, and temporarily disabling the firewall to no avail.  Nothing helped with getting PIO home to startup. 

However, today I accidentally resolved this issue(!!) while trying to fix my python path (python3) and install requirements for another extension.  It turned out that I had installed several different versions of python3 in the past and had not cleaned up the entries in my system path environment variable!  Once I deleted other python install folders in two other places and (probably mainly) deleted the old python path variables, suddenly PIO started happily reinstalling and when I started PIO home, it loaded!

It seems rather 'elementary' now, but take it from me:  make sure your system path to your python3 folder is right, without other leftovers causing installation issues.

Hopefully this helps someone else, as I don't think anyone has posted this particular cause yet...maybe I missed it if so.

chrissivo, 2022-11-04:
Hi @rocketfuze ,

Thanks for posting your solution. I am working on a similar device (Windows 10, corporate network), tried out the different hints, but didn't succeed so far. So i deleted a Python 3.10-installation and when opening VS Code again, PlatformIO was reinstalling again. However, it did not load.

Can you tell me, which version of Python you are using? Is it 3.11.0 or an older one?

rocketfuze, 2022-11-04:
I'm using Python 3.10, but I wouldn't guess that matters.  The (only) python variables I have installed currently in my system path are to the installation folder as follows:

C:\Users\...\Python\Python310\Scripts
C:\Users\...\Python\Python310

Hope this helps you @chrissivo!

chrissivo, 2022-11-04:
Thanks, @rocketfuze for this information. But no, it didn't. I replaced Python 3.11.0 with Python 3.10.8. I tried to install it for the local user and for everybody. Also changed the environment variables. I opened ports from 8008 to 8100. Nothing helped so far. But thanks for getting back that fast.

chrissivo, 2022-11-04:
Finally I found a solution working for me: https://community.platformio.org/t/ssl-error-platform-io/19920/2

"In VS Code File>Preferences>Settings search for ssl, Application>Proxy Uncheck Proxy Strict SSL."

rocketfuze, 2022-11-04:
Yes @chrissivo, I remember now that I also have to run with that same setting unchecked, but it had been a while since I had gotten it working before with just that, so I didn't think to mention it...apologies for leaving out that important piece!  Glad you were able to get yours working as well.

minorikawafuso, 2022-11-22:
> 

> I was having the same problems and have fixed this on Ubuntu recently upgraded to 21.04:
> 
>     * I had python3.9 installed but platformio was using python3.7 and trying to access some of the libraries from 3.7 which were not accessible.
> 
>     * I have uninstalled the semantic-version using pip 3.9 `pip3.9 uninstall semantic-version`
>       
>       * if Errno13. then just do a `sudo rm -rf /usr/lib/python3/dist-packages/semantic_version*`
> 
>     * Python3.7 on ubuntu 21.04 is broken (libffi does not exist)
> 
>     * Installed [pyenv following these instructions](https://github.com/pyenv/pyenv-installer)
> 
>     * `pyenv install 3.7.11`
> 
>     * `pyenv global 3.7.11`
> 
>     * Installed semantic-version and platformio `pip install semantic-version platformio`
> 
>     * Restart VSCode
> 
>     * Let PlatformIO reinstall itself, now it will finish and open up

I only ran `pip install semantic-version platformio` and it works! thx

[3 more replies]