platformio/platformio-vscode-ide #206: Installation Manager

bartlbyCv, 2018-07-24:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.3
System: Windows_NT, 10.0.15063, x64

# Exception
```
Error: Failed to download file https://files.pythonhosted.org/packages/ae/e8/2340d46ecadb1692a1e455f13f75e596d4eab3d11a57446f08259dee8f02/pip-10.0.1.tar.gz: Error: connect ETIMEDOUT 151.101.1.63:443
	at E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:34588
	at P (E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:15208)
	at Generator._invoke (E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:14996)
	at Generator.e.(anonymous function) [as throw] (E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:15387)
	at n (E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:36247)
	at E:\VSCodePortable\Data\code\extensions\platformio.platformio-ide-0.17.3\node_modules\platformio-node-helpers\lib\index.js:1:36366
	at <anonymous>
	at process._tickCallback (internal/process/next_tick.js:109:7)
```

ivankravets, 2018-07-24:
Could you check your Internet connection?

bartlbyCv, 2018-07-24:
Perhaps this is due to the company's security policy.
Tell me. Is it possible to make it portable or install manually?

ivankravets, 2018-07-24:
Sure, see http://docs.platformio.org/en/latest/installation.html#virtual-environment

bartlbyCv, 2018-07-25:
Спасибо, это очень помогло. Подскажите есть инструкция как вручную установить библиотеки и платы?

ivankravets, 2018-07-26:
http://docs.platformio.org/en/latest/userguide/index.html

closed 2018-07-26 by ivankravets