platformio/platformio-vscode-ide #222: Installation Manager

sclub, 2018-07-31:
# Description of problem
Installation fails, regardless what I have done.
Already removed python2.7 and python3 installs, nothing helps.

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.24.0
PIO IDE: v0.17.4
System: Windows_NT, 10.0.17134, ia32

# Exception
```
PIP: C:\Users\s.club\.platformio\penv\Scripts\python.exe: No module named pip

 If you have antivirus software in a system, try to disable it for a while.
```

ivankravets, 2018-08-03:
Could you try to remove this folder `C:\Users\s.club\.platformio\penv` and restart?

sclub, 2018-08-06:
After removing my "normal" Python for testing purposes it works.
So if anyone runs into that problem it might be a good idea to clean up all your python stuff and start over again.

closed 2018-08-06 by sclub