platformio/platformio-vscode-ide #226: C%2FC%2B%2B%20project%20index%20rebuild%20failed

wolfluiz, 2018-08-02:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-atom-ide%2Fissues%0A%0A%23%20Configuration%0AAtom%3A%201.29.0%0APIO%20IDE%3A%20v2.1.12%0ASystem%3A%20Windows_NT%2C%2010.0.16299%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Traceback%20(most%20recent%20call%20last)%3A%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cplatformio%5C__main__.py%22%2C%20line%20120%2C%20in%20main%0D%0A%20%20%20%20cli(None%2C%20None%2C%20None)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Ccore.py%22%2C%20line%20700%2C%20in%20__call__%0D%0A%20%20%20%20return%20self.main(*args%2C%20**kwargs)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Ccore.py%22%2C%20line%20680%2C%20in%20main%0D%0A%20%20%20%20rv%20%3D%20self.invoke(ctx)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Ccore.py%22%2C%20line%201027%2C%20in%20invoke%0D%0A%20%20%20%20return%20_process_result(sub_ctx.command.invoke(sub_ctx))%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Ccore.py%22%2C%20line%20873%2C%20in%20invoke%0D%0A%20%20%20%20return%20ctx.invoke(self.callback%2C%20**ctx.params)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Ccore.py%22%2C%20line%20508%2C%20in%20invoke%0D%0A%20%20%20%20return%20callback(*args%2C%20**kwargs)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cclick%5Cdecorators.py%22%2C%20line%2016%2C%20in%20new_func%0D%0A%20%20%20%20return%20f(get_current_context()%2C%20*args%2C%20**kwargs)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cplatformio%5Ccommands%5Cinit.py%22%2C%20line%20102%2C%20in%20cli%0D%0A%20%20%20%20pg%20%3D%20ProjectGenerator(project_dir%2C%20ide%2C%20env_name)%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cplatformio%5Cide%5Cprojectgenerator.py%22%2C%20line%2035%2C%20in%20__init__%0D%0A%20%20%20%20self._gather_tplvars()%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cplatformio%5Cide%5Cprojectgenerator.py%22%2C%20line%20151%2C%20in%20_gather_tplvars%0D%0A%20%20%20%20self._tplvars.update(self.get_project_build_data())%0D%0A%20%20File%20%22c%3A%5Cusers%5Cluiz%20henrique%5C.platformio%5Cpenv%5Clib%5Csite-packages%5Cplatformio%5Cide%5Cprojectgenerator.py%22%2C%20line%2075%2C%20in%20get_project_build_data%0D%0A%20%20%20%20raise%20result.exception%0D%0AIOError%3A%20%5BErrno%202%5D%20No%20such%20file%20or%20directory%3A%20'C%3A%5C%5CUsers%5C%5Cluiz%20henrique%5C%5CDocuments%5C%5Cdocumetos%20motemper%5C%5Ctrofeus%20botocudos%5C%5CMarlin-bugfix-2.0.x%5C%5CMarlin-bugfix-2.0.x%5C%5C.piolibdeps%5C%5C_tmp_installing-xvjzfa-package%5C%5CTMC26XStepper-c1921b4856a5ef4b659b661b8cb5dd56f3075e57%5C%5Cdocumentation%5C%5Chtml%5C%5C_t_m_c26_x_stepper_8cpp_source.html'%0D%0A%0D%0A%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%0D%0A%0D%0AAn%20unexpected%20error%20occurred.%20Further%20steps%3A%0D%0A%0D%0A*%20Verify%20that%20you%20have%20the%20latest%20version%20of%20PlatformIO%20using%0D%0A%20%20%60pip%20install%20-U%20platformio%60%20command%0D%0A%0D%0A*%20Try%20to%20find%20answer%20in%20FAQ%20Troubleshooting%20section%0D%0A%20%20http%3A%2F%2Fdocs.platformio.org%2Fpage%2Ffaq.html%0D%0A%0D%0A*%20Report%20this%20problem%20to%20the%20developers%0D%0A%20%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-core%2Fissues%0D%0A%0D%0A%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%3D%0D%0A%0D%0A%0A%60%60%60%0A

ivankravets, 2018-08-03:
Please check your `platformio.ini` and that it is not broken.

closed 2018-08-03 by ivankravets

wolfluiz, 2018-08-16:
It's just bad installation the clang c things