platformio/platformio-vscode-ide #229: MSVC Runtime Error when starting PIO

tovine, 2018-08-03:
[image]

This error appears when starting PIO, however after dismissing it I can still seem to browse the PIO menus just fine.

System info:
- Windows 7 (64-bit)
- Python 2.7
- Fresh install of VS Code (with only language extensions for Python and C++ enabled)

ivankravets, 2018-08-03:
Could you try manually create virtual environment for PIo Core? http://docs.platformio.org/en/latest/installation.html#virtual-environment

ivankravets, 2018-08-09:
Could you try this solution? http://docs.platformio.org/en/latest/faq.html#importerror-cannot-import-name-remove-dead-weakref

tovine, 2018-08-09:
That worked, problem solved!

closed 2018-08-09 by tovine