platformio/platformio-vscode-ide #242: C%2FC%2B%2B%20project%20index%20rebuild%20failed

nomisf, 2018-08-07:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-atom-ide%2Fissues%0A%0A%23%20Configuration%0AAtom%3A%201.29.0%0APIO%20IDE%3A%20v2.1.12%0ASystem%3A%20Windows_NT%2C%2010.0.17134%2C%20ia32%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Processing%20efm32hg_stk3400%20(platform%3A%20siliconlabsefm32%3B%20board%3A%20efm32hg_stk3400%3B%20framework%3A%20arduino)%0D%0A%0D%0AVerbose%20mode%20can%20be%20enabled%20via%20%60-v%2C%20--verbose%60%20option%0D%0APLATFORM%3A%20Silicon%20Labs%20EFM32%20%3E%20Silicon%20Labs%20SLSTK3400A%20USB-enabled%20(Happy%20Gecko)%0D%0ASYSTEM%3A%20EFM32HG322F64%2024MHz%208KB%20RAM%20(64KB%20Flash)%0D%0ADEBUG%3A%20CURRENT(jlink)%20ON-BOARD(jlink)%20EXTERNAL(blackmagic)%0D%0AError%3A%20This%20board%20doesn't%20support%20arduino%20framework!%0D%0A%20%5BERROR%5D%20Took%201.12%20seconds%20%0D%0A%0D%0A%0A%60%60%60%0A

nomisf, 2018-08-07:
Never mind, this must be mbed and not arduino.

ivankravets, 2018-08-10:
Yes, this board does not support arduino framework.

closed 2018-08-10 by ivankravets