platformio/platformio-vscode-ide #255: C%2FC%2B%2B%20project%20index%20rebuild%20failed

gangss2, 2018-08-10:
%23%20Description%20of%20problem%0ALeave%20a%20comment...%0A%0ABEFORE%20SUBMITTING%2C%20PLEASE%20SEARCH%20FOR%20DUPLICATES%20IN%0A-%20https%3A%2F%2Fgithub.com%2Fplatformio%2Fplatformio-atom-ide%2Fissues%0A%0A%23%20Configuration%0AAtom%3A%201.29.0%0APIO%20IDE%3A%20v2.1.12%0ASystem%3A%20Windows_NT%2C%206.1.7601%2C%20x64%0A%0A%23%20Exception%0A%60%60%60%0AError%3A%20Processing%20nanoatmega328%20(platform%3A%20atmelavr%3B%20board%3A%20nanoatmega328%3B%20framework%3A%20arduino)%0D%0A%0D%0AVerbose%20mode%20can%20be%20enabled%20via%20%60-v%2C%20--verbose%60%20option%0D%0APLATFORM%3A%20Atmel%20AVR%20%3E%20Arduino%20Nano%20ATmega328%0D%0ASYSTEM%3A%20ATMEGA328P%2016MHz%202KB%20RAM%20(30KB%20Flash)%0D%0AConverting%20ADS1115_single.ino%0D%0ALibrary%20Dependency%20Finder%20-%3E%20http%3A%2F%2Fbit.ly%2Fconfigure-pio-ldf%0D%0ALDF%20MODES%3A%20FINDER(chain)%20COMPATIBILITY(soft)%0D%0ACollected%2049%20compatible%20libraries%0D%0AScanning%20dependencies...%0D%0AError%3A%20Could%20not%20find%20%60I2Cdevlib-Core%60%20dependency%20for%20%60I2Cdevlib-ADS1115%60%20library%0D%0A%20%5BERROR%5D%20Took%202.60%20seconds%20%0D%0A%0D%0A%0A%60%60%60%0A

ivankravets, 2018-08-11:
Check internet connection. See https://platformio.org/lib/search?query=framework%253Aarduino%2520name%253AI2Cdevlib-Core&page=1

ivankravets, 2018-08-11:
```
 pio lib -g install "I2Cdevlib-ADS1115"
Library Storage: /Users/ikravets/.platformio/lib
Looking for I2Cdevlib-ADS1115 library in registry
Found: https://platformio.org/lib/show/95/I2Cdevlib-ADS1115
LibraryManager: Installing id=95
Downloading...
Unpacking  [####################################]  100%
I2Cdevlib-ADS1115 @ 3b4c8bda90 has been successfully installed!
Installing dependencies
Looking for I2Cdevlib-Core library in registry
Found: https://platformio.org/lib/show/11/I2Cdevlib-Core
LibraryManager: Installing id=11
Downloading...
Unpacking  [####################################]  100%
I2Cdevlib-Core @ 615f132009 has been successfully installed!
```

closed 2018-08-11 by ivankravets