platformio/platformio-vscode-ide #257: IntelliSense Index Rebuild

laurentopia, 2018-08-10:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.25.1
PIO IDE: v0.17.4
System: Windows_NT, 10.0.14393, x64

# Exception
```
Error: Processing d1_mini (platform: espressif32; board: d1_mini; framework: arduino)

PlatformManager: Installing espressif32
Downloading...
Unpacking...
espressif32 @ 1.2.0 has been successfully installed!
The platform 'espressif32' has been successfully installed!
The rest of packages will be installed automatically depending on your build environment.
PackageManager: Installing toolchain-xtensa32 @ ~2.50200.0
Downloading...
Unpacking...
PackageManager: Installing framework-arduinoespressif32 @ ~1.10000.0
Downloading...
Unpacking...
PackageManager: Installing tool-esptoolpy @ ~1.20500.0
Downloading...
Unpacking...
Verbose mode can be enabled via `-v, --verbose` option
Error: Unknown board ID 'd1_mini'
 [ERROR] Took 27.28 seconds 

```

ivankravets, 2018-08-11:
`d1_mini` is based on ESP8266. See http://docs.platformio.org/en/latest/platforms/embedded_boards.html#wemos

closed 2018-08-11 by ivankravets

laurentopia, 2018-08-12:
solution was to re import the project that was done on esp32 (d1 mini) with d1 r32 profile