platformio/platformio-vscode-ide #263: serial.serialutil.SerialException: could not open port 'COM5': WindowsError(5, 'Access is denied.')

laurentopia, 2018-08-12:
With a freshly launched vscode, first upload works
subsequent uploads fail even after cleaning
console dump:
`raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port 'COM5': WindowsError(5, 'Access is denied.')
*** [upload] Error 1
`

hardware: d1 r32 (esp32) w/ ch240

Because restarting vscode each time solves the issue for one upload, it's unlikely to be caused by code running on the esp32 but in any case I'll post the code here
`#include <WiFi.h>
#include <WiFiClient.h> 
#include <WebServer.h>
#include <Wire.h>
#include "SSD1306Wire.h"
// #include <Adafruit_MCP23017.h>

const int OLED_SDA=32, OLED_SCL=33, MCP_SDA=32, MCP_SCL=33;

const char* ssid = "<3 2.4";
const char* password = "love8boubou";

int screenW = 128;
int screenH = 64;
int centerX = screenW/2;
int centerY = (screenH/2);
SSD1306Wire display(0x3c, OLED_SDA,OLED_SCL);
void initDisplay()
{
	display.init();
	display.flipScreenVertically();
	display.clear();
}

#pragma region SERVER
void connectWIFI()
{
	display.setTextAlignment(TEXT_ALIGN_CENTER);
	display.setColor(WHITE);
	display.setFont(ArialMT_Plain_10);
	display.drawString(centerX, centerY, "Connecting to Wi-Fi");
	display.display();
	Serial.println("Connecting to Wi-Fi");
	WiFi.mode(WIFI_STA);
	WiFi.begin (ssid, password);
	int waitCounter=0;
	while (WiFi.status() != WL_CONNECTED) {
		Serial.print(".");
		display.drawProgressBar(20, centerY+20, 128-40, 8, waitCounter);
		waitCounter++;
		waitCounter %=101;
		display.display();
		delay(100);
	}
	Serial.println("success!");
	Serial.print("IP Address is: ");
	Serial.println(WiFi.localIP());
}

// ------------ SERVER

WebServer server(80);
int LDRPin = 5;
int LDRReading = 0; 
int milisInterval = 2000;
int count = 0;

void handleRoot() {
    LDRReading = analogRead(LDRPin);
[... 238 lines cut ...]
void MCP_digitalWrite(uint8_t pin, uint8_t d) {
	uint8_t gpio;
	uint8_t bit=bitForPin(pin);

	// read the current GPIO output latches
	uint8_t regAddr=regForPin(pin,MCP23017_OLATA,MCP23017_OLATB);
	gpio = readRegister(regAddr);

	// set the pin and direction
	bitWrite(gpio,bit,d);

	// write the new GPIO
	regAddr=regForPin(pin,MCP23017_GPIOA,MCP23017_GPIOB);
	writeRegister(regAddr,gpio);
}

void MCP_pullUp(uint8_t p, uint8_t d) {
	updateRegisterBit(p,d,MCP23017_GPPUA,MCP23017_GPPUB);
}

uint8_t MCP_digitalRead(uint8_t pin) {
	uint8_t bit=bitForPin(pin);
	uint8_t regAddr=regForPin(pin,MCP23017_GPIOA,MCP23017_GPIOB);
	return (readRegister(regAddr) >> bit) & 0x1;
}`

ivankravets, 2018-08-30:
It seems that COM port is opened in another monitor. Try to reconnect board.

closed 2018-08-30 by ivankravets

bharath5673, 2019-06-09:
may be the alignment error or indentation error in the script... double check those lines ...

SergeyShatsky, 2021-03-05:
Try to change USB cable