platformio/platformio-vscode-ide #279: Start PIO Home Server

thinkfire, 2018-08-16:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-atom-ide/issues

# Configuration
Atom: 1.29.0
PIO IDE: v2.1.13
System: Windows_NT, 10.0.17134, x64

# Exception
```
'platformio' is not recognized as an internal or external command,
operable program or batch file.
Error: spawn platformio ENOENT
```

ivankravets, 2018-08-16:
Could you reproduce this issue after restart of VSCode? When do you see this error? Would be thankful for any details.

thinkfire, 2018-08-18:
actually, I installed everything once again virtual environment and now its working properly.

ivankravets, 2018-08-30:
Thanks for the report!

closed 2018-08-30 by ivankravets