platformio/platformio-vscode-ide #296: Installation Manager

jgracio, 2018-08-22:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.26.1
PIO IDE: v0.17.5
System: Windows_NT, 10.0.17134, x64

# Exception
```
PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio/archive/develop.zip (Caused by SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),))

 If you have antivirus software in a system, try to disable it for a while.
```

ivankravets, 2018-08-30:
Try to uninstall and install the latest version of Python 2.7

closed 2018-08-30 by ivankravets

jgracio, 2018-08-30:
Thanks
The latest version is already installed 2.7.15. Any other tip?

jgracio, 2018-08-30:
Thanks again but error again. This time I kept my Kaspersky enabled

PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio/archive/develop.zip (Caused by SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),))

 If you have antivirus software in a system, try to disable it for a while.

ivankravets, 2018-08-30:
Could you try to navigate to Windows > Control Panel > Programs > Remove ALL Pythons from a system. Then, remove again `C:\Users\YourUserName.platformio\penv ` folder. Something is wrong with your Python

jgracio, 2018-08-30:
I removed all python items from programs and again the penv file., closed the vb and restarted it again. No reusult:

PIP:   Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
  Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),)': /platformio/platformio/archive/develop.zip
Could not install packages due to an EnvironmentError: HTTPSConnectionPool(host='github.com', port=443): Max retries exceeded with url: /platformio/platformio/archive/develop.zip (Caused by SSLError(SSLError(1, u'[SSL: CERTIFICATE_VERIFY_FAILED] certificate verify failed (_ssl.c:726)'),))

 If you have antivirus software in a system, try to disable it for a while.

ivankravets, 2018-08-30:
Please provide an output from system Terminal `cmd` (not from PIO IDE)
```
echo %PATH%
echo %PYTHONPATH%
python --version
pip --version
pip list
```

jgracio, 2018-08-30:
C:\Users\admin>echo %PATH%
C:\ProgramData\Oracle\Java\javapath;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\Skype\Phone\;C:\Program Files (x86)\QuickTime\QTSystem\;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\Microsoft VS Code\bin;C:\Users\admin\AppData\Local\Microsoft\WindowsApps;C:\Users\admin\AppData\Local\GitHubDesktop\bin;%USERPROFILE%\AppData\Local\Microsoft\WindowsApps;

The remaining coms are returning errors ... I suppose because Python was uninstalled

jgracio, 2018-08-31:
Please don't give up on me |:

ivankravets, 2018-08-31:
Please try the last option. Again, navigate to Control Panel > Remove All Pythons. Now, use step-by-step this instruction http://docs.platformio.org/en/latest/installation.html#virtual-environment

Does it work now?

ivankravets, 2018-08-31:
Do not forget to install Python manually using http://docs.platformio.org/en/latest/faq.html#faq-install-python

jgracio, 2018-08-31:
Thanks you again~.. some progress
I did as per your instruction up to run the command "pip install -U https://github.com/platformio/platformio-core/archive/develop.zip" getting the same errors as through the vbc. Then I downloaded the file from github and ran the same command but now pointing to the temporary directory in my drive and all went without errors.

Then I restarted  vbc but got exactly the same error messages as before because it tried to install platform-core again from github.

Do you see this as being closer to a happier end?

ivankravets, 2018-08-31:
Please mail me@ikravets.com directly. I will provide some hints.

jgracio, 2018-08-31:
Great