platformio/platformio-vscode-ide #306: Installation Manager

AvatarWeb1, 2018-08-26:
# Description of problem
Leave a comment...

BEFORE SUBMITTING, PLEASE SEARCH FOR DUPLICATES IN
- https://github.com/platformio/platformio-vscode-ide/issues

# Configuration

VSCode: 1.26.1
PIO IDE: v0.17.5
System: Windows_NT, 10.0.17134, x64

# Exception
```
Conda Virtualenv: 'conda' n'est pas reconnu en tant que commande interne
ou externe, un programme ex�cutable ou un fichier de commandes.
Error: spawn conda ENOENT
```

ivankravets, 2018-08-30:
Please install PIO Core in virtual environment manually http://docs.platformio.org/en/latest/installation.html#virtual-environment

closed 2018-08-30 by ivankravets

AvatarWeb1, 2018-10-07:
Merci Beaucoup @AvatarWeb1 #358 
[image]