platformio/platformio-vscode-ide #319: PlatformIO does not start automatically form time to time after openning (re)starting VSCode

Navis-Raven, 2018-09-03:
What kind of issue is this?

- [X] **PlatformIO Core**.
      If you’ve found a bug, please provide an information below.

------------------------------------------------------------------

### Configuration

**Operating system**: Windows 8.1 (and it occurs also on Windows 7)

**PlatformIO Version** (`platformio --version`): 3.6.1a1 on VSCode

### Description of problem
Since the new platformIO interface on VSCode there is a problem I first saw it in windows 7 an then also in windows 8.1

It's a random bug: When starting VSCode the platformIO core should start as well, however from time to tome it does not start. And it particulary happens when creating a new project, and restart VSCode, see the image belon

[image]

As you can see there is no PIO logo as nor PIO toolbar. To have this screenshot, I've started a New Project, type the name of the project and selected ES32 Wrover Kit, and a few seconds later, vscode restarted with my new project but without pio toolbar.

DMDallas, 2018-09-05:
Same problem. 
Configuration
Operating system: Windows Windows 7
PlatformIO Version (platformio --version): 3.6.1a1 on VSCode

Navis-Raven, 2018-09-07:
I had again the same bug, but this time with this message

[image]

closed 2018-09-07 by ivankravets

Navis-Raven, 2018-09-08:
It seems that Ivan has fixed the bug.

commit cd369081 (2018-09-07): Fixed issue when extension does not start after adding new project folder to workspace // Resolve #319