platformio/platformio-vscode-ide #322: PIO Home don't load and don't accept commands

Ircbarros, 2018-09-05:
### **CONFIGURATION**
**Operating System:** Windows 10 v. 1803 (OS 17134.228)
**PlatformIO Version:** 3.6.1a1

_Command via CMD_

**ECHO %PATH%**: C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\system32\Wbem;C:\WINDOWS\system32\WindowsPowerShell\v1.0\;C:\ProgramData\Anaconda3;C:\ProgramData\Anaconda3\Library\mingw-w64\bin;C:\ProgramData\Anaconda3\Library\bin;C:\ProgramData\Anaconda3\Scripts;C:\ProgramData\Anaconda3\envs;C:\Program Files (x86)\Common Files\Oracle\Java\javapath;C:\Program Files (x86)\Intel\iCLS Client\;C:\Program Files\Intel\iCLS Client\;C:\ProgramData\Oracle\Java\javapath;C:\windows\system32;C:\windows;C:\windows\System32\Wbem;C:\windows\System32\WindowsPowerShell\v1.0\;C:\WINDOWS\system32;C:\WINDOWS;C:\WINDOWS\System32\Wbem;C:\WINDOWS\System32\WindowsPowerShell\v1.0\;C:\Program Files (x86)\NVIDIA Corporation\PhysX\Common;C:\Program Files\MATLAB\R2017b\runtime\win64;C:\Program Files\MATLAB\R2017b\bin;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files\Intel\Intel(R) Management Engine Components\DAL;C:\Program Files (x86)\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files\Intel\Intel(R) Management Engine Components\IPT;C:\Program Files (x86)\Xoreax\IncrediBuild;C:\Program Files\Microsoft SQL Server\120\Tools\Binn\;C:\Program Files\Common Files\Autodesk Shared\;C:\Program Files\Git\cmd;C:\WINDOWS\System32\OpenSSH\;C:\Program Files\nodejs\;C:\Program Files\Zerynth\ztc\windows64;C:\Program Files\kubectl;C:\Program Files\helm\windows-amd64;C:\Program Files\IBM\Cloud\bin;C:\SysGCC\esp8266\bin;C:\ProgramData\Anaconda2;C:\Users\Ítalo Renan\AppData\Local\Programs\Microsoft VS Code\bin

**echo %PYTHONPATH%:** %PYTHONPATH%

_Command via PIO Terminal_

**python --version:**Python 2.7.15 :: Anaconda, Inc. (command via PIO Terminal)
**pip --version:** pip 18.0 from c:\.platformio\penv\lib\site-packages\pip (python 2.7)
**pip list:**
Package          Version
---------------- ---------
bottle           0.12.13
certifi          2018.8.24
chardet          3.0.4
click            5.1
colorama         0.3.9
idna             2.7
pip              18.0
platformio       3.6.1a1
pyserial         3.4
requests         2.19.1
semantic-version 2.6.0
setuptools       40.2.0
urllib3          1.23
wheel            0.31.1
wincertstore     0.2

### **ISSUES**

- The homepage does not work (PIO Home don't load)
- Some commands don't work, like platformiobuild or platformiotest, but the terminal seems to work fine
- The >pio home command gives a cannot /GET, but the >pio home --8888 works
- Can't activate an environment via conda command

### **ALREADY TRIED**

- Uninstall/Install PIO Core and removed %HOME_DIR%/.platformio/penv
- Uninstall/Install Anaconda/Miniconda
- Uninstall/Install Python

[image]

Navis-Raven, 2018-09-07:
it looks similar to my issue: https://github.com/platformio/platformio-vscode-ide/issues/319

ivankravets, 2018-09-07:
1. Please open windows control center, installed software. Remove all Python from a system.
2. Remove `C:\Users\***\.platformio\penv` and `C:\Users\***\.platformio\python27` folders if exist
3. Update extension in VScode to 1.1.1

Does it work now?

closed 2018-09-10 by ivankravets