platformio/platformio-vscode-ide #328: Python binary included in ~/.platformio/penv/bin does not work on Fedora 28

ciffer, 2018-09-11:
# Description of problem
After following steps described at https://docs.platformio.org/en/latest/installation.html#method-1 following command fails:
```
$ pio
zsh: /home/ciffer/.platformio/penv/bin/pio: bad interpreter: /home/ciffer/.platformio/penv/bin/python2.7: no such file or directory
```
Turns out that the file actually exists:
```
$ file /home/ciffer/.platformio/penv/bin/python2.7
/home/ciffer/.platformio/penv/bin/python2.7: ELF 64-bit LSB executable, x86-64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-x86-64.so.2, for GNU/Linux 3.2.0, BuildID[sha1]=0df3d552e4c08413f06c13955c17c2764e1dc81c, stripped
```
After some digging I learned that `/lib/ld-linux-x86-64.so.2` lib is missing but I found it at `/lib64/ld-linux-x86-64.so.2`. Creating symbolic link makes `pio` work again:
```
$ sudo ln -s /lib64/ld-linux-x86-64.so.2 /lib/ld-linux-x86-64.so.2
$ pio                                                             

* regular pio output here*
```

Removing `~/.platformio/penv/bin/python2.7` and linking it to `$(which python2.7)` works fine as well
Note that installing PlatformIO via `sudo pip install -U platformio` is also OK

# Expected result
After following steps described at https://docs.platformio.org/en/latest/installation.html#method-1 `pio` command should work fine without any system lib or pio modifiactions

# Configuration
Fedora release 28 (Twenty Eight)
VSCode 1.27.1
PlatformIO: Home 1.0.2, Core 3.6.0

ivankravets, 2018-09-21:
Does it work if you do?
```
$ source /home/ciffer/.platformio/penv/bin/activate
$ (penv) pio
```

ciffer, 2018-09-21:
No. It gives the same error

ivankravets, 2018-09-21:
Which version of `python-virtualenv` do you use? it does not create a valid virtual environment. 

Maybe, report to https://github.com/pypa/virtualenv ?

closed 2018-10-26 by ivankravets