platformio/platformio-vscode-ide #330: Start PIO Home Server

heepdog, 2018-09-12:
# Description of problem

After opening VS the PIO server is unable to start and throws an import error.  This is the second time trying to upgrade PIO from a previous version.  The first time, an error corresponding to issue #142.  A fix for that was to delete the home folder and reinstall, which I did.  The install seemed to go fine but when PIO loads, it cannot start the server and throws this error.

# Configuration

VSCode: 1.27.1
PIO IDE: v1.1.1
System: Windows_NT, 10.0.17744, x64

# Exception
```
Traceback (most recent call last):
  File "main.py", line 13, in init main
  File "c:\users\chris\.platformio\penv\lib\site-packages\platformio\app.py", line 25, in <module>
    import requests
  File "c:\users\chris\.platformio\penv\lib\site-packages\requests\__init__.py", line 43, in <module>
    import urllib3
  File "c:\users\chris\.platformio\penv\lib\site-packages\urllib3\__init__.py", line 8, in <module>
    from .connectionpool import (
  File "c:\users\chris\.platformio\penv\lib\site-packages\urllib3\connectionpool.py", line 29, in <module>
    from .connection import (
  File "c:\users\chris\.platformio\penv\lib\site-packages\urllib3\connection.py", line 39, in <module>
    from .util.ssl_ import (
  File "c:\users\chris\.platformio\penv\lib\site-packages\urllib3\util\__init__.py", line 6, in <module>
    from .ssl_ import (
  File "c:\users\chris\.platformio\penv\lib\site-packages\urllib3\util\ssl_.py", line 4, in <module>
    import hmac
  File "C:\Users\Chris\.platformio\python27\Lib\hmac.py", line 8, in <module>
    from operator import _compare_digest as compare_digest
ImportError: cannot import name _compare_digest

```

ivankravets, 2018-09-17:
Please reopen if you still need help

closed 2018-09-17 by ivankravets

heepdog, 2018-09-17:
I just attempted the fix from #157.  When I went to uninstall the python versions I had installed, there were 8 different programs related to python.  

[image]
Since PIO used a 2.7.X version, I removed just those .  After opening up VScode, PIO was reinstalled and it seems that it is working properly.