platformio/platformio-vscode-ide #335: Update tasks are missing

sstaub, 2018-09-15:
After the last update you bring back the Run Task... button, but you delete the update tasks from the task list, annoying.
It would be nice if can bring back the old style init procedure with tasks, before PIO home.

sstaub, 2018-09-16:
In the task list I'm missing the update tasks. which are deleted after last update.
That is most annoying to change workflow after each update.

ivankravets, 2018-09-17:
We had an issue with updates related to libraries. The previous version supported only updating of global libraries. That was the problem for people. Now, we refactored all tasks and made them behave in a classic VSCode workflow: Tasks Per project.

So, the global actions (not tasks) were moved to "Quick Access" activity view. See
[image]

I know that it's bad when something changes. We want to clean up all the misunderstands from our UI.

sstaub, 2018-09-17:
My personal problem is that "Platform updates" opens first PIO Home tab instead of a simple task
using the terminal.
Meanwhile I get back to use the CLI, so 'pio update' does it for me.

ivankravets, 2018-09-17:
We left this command in VSCode global scope. Just Shift+CMD+P => platformio update. See
[image]

sstaub, 2018-09-17:
It would we nice to have it also in the task list with dedicated button.

sstaub, 2018-09-21:
Solved, but can you add it for the task list opend by tasks button.

commit fbfca74d (2018-09-21): Add "Update All" (platforms, packages, libraries) quick access command to "PlatformIO Activity" (left sidebar) // Resolve #335
commit ebdc1fbf (2018-10-02): Add "Update All (libraries, platforms, and packages)" and "Upgrade PlatformIO Core" tasks to global VSCode Task Manager // Resolve #335