platformio/platformio-vscode-ide #346: PIO home tab gets back to the home page when losing focus

Macfly, 2018-09-25:
### Description of problem
when leaving and going back to PIO home tab, it gets back to home. It does not keep the state.
For example:
I'm on the PIO home tab browsing the examples for the Espressif 32 platform. I click on a source file in my projet (without closing PIO Home tab).
When I got back to the PIO Home tab, I'm back at the home screen.
Below a gif of the issue

### Configuration
VSCode: 1.28.0 insider
PIO IDE: v1.3.0
Core: 3.6.1a4
System: Windows 10 x64

[image]

ivankravets, 2018-09-26:
I can't reproduce this issue with 1.27. Could you reproduce it with 1.27 and after VScode restart?

Macfly, 2018-09-27:
yes it works with 1.27.2, so I guess I'll stop using the insider until they fix it

ivankravets, 2018-09-27:
Oh, I think it's a bug in the VSCode.

@mjbvz Matt, did you change something in WebView API?

ivankravets, 2018-10-02:
@Macfly I've just installed the latest insider build which is planned for release tomorrow and it totally breaks official VSCode documentation as for the Webview API. 

I opened a separate issue for that: https://github.com/Microsoft/vscode/issues/59823

I recommend staying with VSCode 1.27 while the next bug fix release for 1.28.

Macfly, 2018-10-02:
I don't have this issue anymore with the version:
Version: 1.28.0-insider (system setup)
Commit: 2d676e6dfdb99bd3e079ac3f4e9833656fc1b607
Date: 2018-10-02T20:44:40.713Z
Electron: 2.0.9
Chrome: 61.0.3163.100
Node.js: 8.9.3
V8: 6.1.534.41
Architecture: x64

ivankravets, 2018-10-03:
Yes, it was fixed in https://github.com/Microsoft/vscode/issues/59823

We would be thankful for the further reports if PlatformIO extension does not work with VSCode Insider.

closed 2018-10-03 by ivankravets