platformio/platformio-vscode-ide #347: Post Build scripts are building before the build finished.

Protoneer, 2018-09-27:
I created a pre and post script to show when they are running. When I clean out my .pioenvs folder the build fails because the "Post" script it is trying to grab a build file before its been created. (Post script running before build happend)

**platformio.ini**
`[env:esp32doit-devkit-v1]`
`platform = espressif32`
`board = esp32doit-devkit-v1`
`framework = arduino`
`upload_port = 10.82.1.8`
`monitor_speed = 115200`
`extra_scripts =`
`  pre:scripts/preBuild.py`
`  post:scripts/postBuild.py`

Output from build:
> Executing task: C:\Users\ask\.platformio\penv\Scripts\platformio.exe run <

Processing esp32doit-devkit-v1 (framework: arduino; platform: espressif32; board: esp32doit-devkit-v1)
--------------------------------------------------------------------------------------------------------------
Verbose mode can be enabled via `-v, --verbose` option
**Pre build scripts**
PLATFORM: Espressif 32 > DOIT ESP32 DEVKIT V1
SYSTEM: ESP32 240MHz 320KB RAM (4MB Flash)
DEBUG: CURRENT(esp-prog) EXTERNAL(esp-prog, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny)
Library Dependency Finder -> http://bit.ly/configure-pio-ldf
LDF MODES: FINDER(chain) COMPATIBILITY(soft)
Collected 30 compatible libraries
Scanning dependencies...
Dependency Graph
|-- <WifiManager> 1.0.0
|   |-- <WiFi> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WebServer> 1.0
|   |   |-- <FS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <DNSServer> 1.1.0
|   |   |-- <WiFi> 1.0
|-- <Update> 1.0
|-- <SPIFFS> 1.0
|   |-- <FS> 1.0
|-- <WebServer> 1.0
|   |-- <FS> 1.0
|   |-- <WiFi> 1.0
|-- <ESPmDNS> 1.0
|   |-- <WiFi> 1.0
|-- <ArduinoOTA> 1.0
|   |-- <ESPmDNS> 1.0
|   |   |-- <WiFi> 1.0
|   |-- <WiFi> 1.0
|   |-- <Update> 1.0
|-- <WiFi> 1.0
|-- <app>
|   |-- <Wire> 1.0
|   |-- <Adafruit_ADS1X15>
|   |   |-- <Wire> 1.0
|   |-- <WebSockets> 2.1.2
|   |   |-- <SPI> 1.0
|   |   |-- <WiFi> 1.0
|   |   |-- <WiFiClientSecure> 1.0
|   |   |   |-- <WiFi> 1.0
|-- <WebSockets> 2.1.2
|   |-- <SPI> 1.0
|   |-- <WiFi> 1.0
|   |-- <WiFiClientSecure> 1.0
|   |   |-- <WiFi> 1.0
|-- <DNSServer> 1.1.0
|   |-- <WiFi> 1.0
**Post build scripts**
False
d:\Code\_deleteMe\ESP32-Power/build/29.bin
IOError: [Errno 2] No such file or directory: 'd:\\Code\\_deleteMe\\ESP32-Power/.pioenvs/esp32doit-devkit-v1/firmware.bin':
File "C:\users\ask\.platformio\penv\lib\site-packages\platformio\builder\main.py", line 175:
env.SConscript(item, exports="env")
File "C:\Users\ask\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 541:
return _SConscript(self.fs, *files, **subst_kw)
File "C:\Users\ask\.platformio\packages\tool-scons\script\..\engine\SCons\Script\SConscript.py", line 250:
exec _file_ in call_stack[-1].globals
File "D:\Code\_deleteMe\ESP32-Power\scripts\postBuild.py", line 19:
shutil.copy(fname, newFileName)
File "C:\Python27\Lib\shutil.py", line 119:
copyfile(src, dst)
File "C:\Python27\Lib\shutil.py", line 82:

ivankravets, 2018-10-03:
Pre/Post means when PlatformIO launches your scripts when constructs build environment. 

It seems that you need Pre/Post actions => http://docs.platformio.org/en/latest/projectconf/advanced_scripting.html#before-pre-and-after-post-actions

closed 2018-10-03 by ivankravets

Protoneer, 2018-10-03:
Thanks for that...
It took me a while to realize that even after saying 

post:scripts/postBuild.py  in the INI

you still need to hook it up to the appropriate event action(buildprog).

Below code worked for me....

`
Import("env", "projenv")
print "Post build scripts"

def after_build(source, target, env):
    print "after_build"
    # do some actions

env.AddPostAction("buildprog", after_build)
`

jshep321, 2021-01-28:
Hmm... 
@Protoneer , this didn't work for me.  Got the Post-build scripts print but then nothing after compile was completed.
What am I missing here? 
  
```
Import("env", "projenv")
import os
print ("Post build scripts")

def after_build(source, target, env):
        print ("after_build")
# do some actions
        os.system('bash -c "./updateRTDBSW.sh"')

        env.AddPostAction("buildprog", after_build)
```

ivankravets, 2021-01-29:
@jshep321 please provide a simple project to reproduce your issue.

jshep321, 2021-01-29:
@ivankravets my problem became obvious after a sleep and a strong cup of coffee --  whitespace/indention matters in python!

Accidentally indented the line: env.AddPostAction("buildprog", after_build)

So unindenting fixed it.  Above, it was considered part of the function, which was never called of course. :)