platformio/platformio-vscode-ide #349: Start PIO Home Server

jankap, 2018-10-02:
# Description of problem
Fresh install of vscode, v 1.27.2, platformio-ide extension v 1.3.1. 
Purged windows 7 python path, reinstalled python 2.7.15. Can be found by cmd.exe which is currently default terminal in VS Code.
platformio-ide install worked and created folder: C:\Users\user\.platformio

After restarting VS Code, PIO Home does not load. After a while "Error: Detected unknown package 'tool-pioplus'" popup occurs.

I assume it has to to something with proxy environmental variables I believe are set correctly:
```
set http_proxy="http://user:pw@proxyip:port"
set https_proxy=%http_proxy%
set HTTP_PROXY=%http_proxy%
set HTTPS_PROXY=%http_proxy%
```

Any ideas? Thanks.

# Configuration

VSCode: 1.27.2
PIO IDE: v1.3.1
System: Windows_NT, 6.1.7601, x64

# Exception
```
Error: Detected unknown package 'tool-pioplus'

```

ivankravets, 2018-10-03:
You need to set these variables into the system environment globally before starting VSCode. Could you try it?

jankap, 2018-10-03:
Yes of course, that's what I did. They are system wide set and can be seen
by echo. After a reboot, too.

ivankravets, 2018-10-03:
Could you try this test? 
http://docs.python-requests.org/en/master/user/advanced/#proxies

ivankravets, 2018-10-03:
```
$ export HTTP_PROXY="http://10.10.1.10:3128"
$ export HTTPS_PROXY="http://10.10.1.10:1080"

$ python
>>> import requests
>>> requests.get('http://example.org')
```

closed 2018-10-09 by ivankravets